ROBOTIS-GIT/OpenCR #18: Dynamixel 상수를 레지스터명으로 변경하는 작업을 해보려 합니다.

chandong83, 2016-10-29:
현재 코드의 몇몇 레지스터 읽고 쓰는 부분이 상수로 되어 있는
부분을 레지스터 명으로 변경하여 조금 더 코드 코드 가독성을 
올려보려 합니다.
아래처럼 말입니다.

ex >>

this->writeByte(bID, 16, level);

to 

<header file side>
                                                                       //(HEX)    Writing Item            Len   Min   Max       
#define DXL_STATUS_RETURN_LEVEL 16 // 0x10 Status Return Level     1       0       2

<source file side>
this->writeByte(bID, DXL_STATUS_RETURN_LEVEL, level);

혹시 의견이나 다이나믹셀 소스코드가 수정중이시라면
말씀해주세요.
이상입니다.

chcbaram, 2016-10-30:
다이나믹셀 라이브러리는 다이나믹셀의 모든 모델을 지원하기 위한 API정도의 역할만 수행합니다. 따라서 레지스터명 같은 경우 모델에 따라 달라지기에 제 생각에는 다이나믹셀 라이브러리를 사용하는 별도의 모터별 클래스를 만들거나 해서 모델별로 지원하는게 더 좋지 않을까 생각됩니다. 

다이나믹셀 라이브러리 같은경우는 기존의 로보티즈 라이브러리를 포팅한 수준이라 수정방향에 대해서는 로보티즈와의 얘기도 필요할 듯 싶습니다.

chandong83, 2016-10-30:
아하 그렇군요~^^
제가 사용해 본 것이 딱 두가지모델이라
다 동일할 것이라고 생각하고 있었네요.
말씀감사합니다.

closed 2016-10-30 by chandong83