ROBOTIS-GIT/OpenCR #19: analogRead 시 아두이노의 핀과 불이치에 대한 의견

chandong83, 2016-10-29:
현재 analogRead를 하려면 일반 아두이노 핀 번호로는
동작을 안하는 것을 확인했습니다.

혹시 이부분에 대해서

g_Pin2PortMapArray 매칭 배열 변수를 아날로그 전용으로 하나 더 만드는 것은 어떤지요?

OpenCR/1.0.0/variants/OpenCR/variant.cpp
extern const Pin2PortMapArray g_Pin2AnalogPortMapArray[]=
{
    {GPIOA, GPIO_PIN_0,   &hADC3,   ADC_CHANNEL_0, NULL, NO_PWM, NO_EXTI },  // 0 A0
    {GPIOF, GPIO_PIN_10,  &hADC3,   ADC_CHANNEL_8, NULL, NO_PWM, NO_EXTI },  // 1 A1
    {GPIOF, GPIO_PIN_9,   &hADC3,   ADC_CHANNEL_7, NULL,  NO_PWM, NO_EXTI },  // 2 A2
    {GPIOF, GPIO_PIN_8,   &hADC3,   ADC_CHANNEL_6, NULL,  NO_PWM, NO_EXTI },  // 3 A3
    {GPIOF, GPIO_PIN_7,   &hADC3,   ADC_CHANNEL_5, NULL,  NO_PWM, NO_EXTI },  // 4 A4
    {GPIOF, GPIO_PIN_6,   &hADC3,   ADC_CHANNEL_4, NULL,  NO_PWM, NO_EXTI },  // 5 A5

    {NULL , 0          ,  NULL,     NO_ADC        , NULL   ,   NO_PWM       , NO_EXTI }
};

analogRead에서는 위의 변수를 참조한다면 아두이노 PIN과 호환성을 줄 수 있지 않을까 합니다.
그리고 이 변수 이하에 나머지 ADC 포트들을 확장하면 어떨가 라는 생각을 하게 되었습니다.

chcbaram, 2016-10-30:
아날로그 입력은 A0~A5를 사용하지 않나요? 직접 핀번호를 입력하는 예는 보지는 못했던것 같습니다. 아두이노 보드마다 A0~A5가 다른 핀들로 매핑되어 있는것들도 있기때문에, 아날로그입력은 핀번호 보다는 A0~A5를 사용하는게 호환성 측면에서 더 좋지 않을까요?

chcbaram, 2016-10-30:
좀더 찾아보니 아두이노에서 analogRead에서 0~5번을 A0~A5에 해당하는 핀으로 다시 매핑하는 부분이 있네요.

`int analogRead(uint8_t pin)
{
	uint8_t low, high;

#if defined(analogPinToChannel)
#if defined(__AVR_ATmega32U4__)
	if (pin >= 18) pin -= 18; // allow for channel or pin numbers
#endif
	pin = analogPinToChannel(pin);
`

anlogPinToChannel()함수가 사용되는데, 핀입력이 0~5의 입력과 원래 핀 두개를 모두 입력 받을 수 있도록 되어 있네요.
참조해서 저희도 원래 핀입력 그리고 0~5로 입력 받았을때 모두 추가하면 될것 같습니다. 말씀하신데로 테이블을 사용하거나, anlogPinToChannel함수와 같이 변환해주는 함수를 만들어도 될것 같구요.

chcbaram, 2016-10-30:
deveop 브랜치에 variant.h에 0~5번 입력에 대한 매크로와 analogRead에서 해당 매크로를 사용하도록 변경하였습니다. 확인 부탁드립니다. 

https://github.com/ROBOTIS-GIT/OpenCR/commit/c2f2289a0a1fbe8732e1aef350c9eb1b6d68515e

chandong83, 2016-10-30:
코드 잘 봤습니다.
이렇게 또 배우네요 ^^

제 보드가 오늘 상태 이상해서 구동은 못 시켜봤습니다.
조만간 테스트 해보겠습니다.
감사합니다.

closed 2016-10-30 by chandong83