ROBOTIS-GIT/OpenCR #26: Dynamixel Pro Libraries

MatMcT, 2017-01-10:
Hi team,

I'm from the Queensland University of Technology and currently working on a project for the Amazon picking challenge. We have 3 of the Dynamixel Pro L54 Series on order. 

We're designing a system that needs to communicate to multiple types of motors, Dynamixel and some others. The current system is using a Teensy 3.6 using the Arduino IDE. What we would like is to also control the Dynamixel motors using the same Teensy so that we can eliminate the need for using the USB2Dynamixel. 

I've been digging through these repositories and have found the Dynamixel.cpp and Dynamixel.h seem to be exactly what I'm looking for.

Would anyone be able to confirm that this is what we need for our application?  
If so would someone also be able to clarify what this pin is supposed to be: BDPIN_DXL_PWR_EN

Thanks in advance. 
Cheers,
Matt

ROBOTIS-Leon, 2017-01-11:
Hi,

I'm sorry for the late reply.
I see what kind of thing you want to make. Keep going!

Initially, to answer to your question, BDPIN_DXL_PWR_EN is the pin of the OpenCM board and it gives the power to the Dynamixel. 

By the way, I should say that the Dynamixel.cpp and Dynamixel.h must be too hard to be ported with the Arduino system, because it was designed for the use in the OpenCM IDE but not for all arduino boards. It looks like what you said, but when you actually are going to port those two, you will have a hard time. 

We are making kind of DynamixelSDK for Arduino now, but a [temporary file](https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/DynamixelSDK) will perhaps work in the other arduino-available boards but not just the OpenCR board.

Plus, the repository of the DynamixelSDK for Arduino will be opened sooner, so if you have another issues which are related to the Arduino stuffs, it would be glad if you ask there. :)

MatMcT, 2017-01-11:
Hi Leon,

Thanks for the reply. Just to elaborate more, we wish to completely control
the DXL directly using our own controller. The controller we're using will
be using the Rx and TX xonnexted to a MAX485 transceiver.  To configure the
serial we also using Serial3. This controller will be receiving usb serial
commands and passing on DXL commands through Serial3(usart)

Looking at the code you linked me to, it seems like we'll be needing to use
the dynamixel_sdk opencr which happens to use the same Serial3
configuration.

Therefore when Configuring the Port handler, we'll need to use the opencr
instance.

Have I understood this correctly?

Cheers

ROBOTIS-Leon, 2017-01-11:
Yes, all the thing you said are correct.

But we will make the DynamixelSDK for Arduino soon :)

MatMcT, 2017-01-11:
Hi Leon,

Thanks for the info! Very helpful. Do you know when the DynamixelSDK for
Adruino will be finished? The project needs to be completed by February so
I might need to make use of what I have now :)

Cheers,
Matt

ROBOTIS-Leon, 2017-01-12:
Hi,

the DynamixelSDK for Arduino will be released in mid February, I guess.
So it seems like you should not wait it but modify the source yourself for your Arduino.
However, I think the SDK for Arduino has only the minor changes from the DynamixelSDK for OpenCR, so you don't need to hesitate to use that :)

closed 2017-03-31 by ROBOTIS-Leon

sitanshu1, 2017-08-19:
Hello, is the library for the arduino released as said earlier? I am unable to use the open cr library function as I am new to that. can anyone please explain how to use it? Thank you.

ROBOTIS-Leon, 2017-08-21:
@sitanshu1 

Hi,

What is mentioned will be postponed due to the system problem.

We are very sorry for unnotified.

DynamixelSDK for Arduino platform is scheduled to be managed by [DynamixelSDK repo](https://github.com/ROBOTIS-GIT/DynamixelSDK)

We should do some test by using some board that represents Arduino platform.
We will notify release data when it is finished.

Thank you for your patience

MatMcT, 2017-08-21:
Hi All,

I managed to use the Dynamixel SDK which i used with a Teensy with a few
modifications. I can share what i did if anyone is interested.

ROBOTIS-Leon, 2017-08-21:
@MatMcT 

Hey, what a good

Where should I go to

MatMcT, 2017-08-21:
I'll upload it when I get a chance today :)

ROBOTIS-Leon, 2017-08-21:
@MatMcT 

Great

sitanshu1, 2017-08-21:
@MatMcT 
Hey, that will be great. Thanks a lot for the same. Kindly let us know the link as I am unable to do so from a long time, and stuck in a project.

Thank you.

MatMcT, 2017-08-25:
Hi All,

I'm currently writing a paper for ICRA and would like to give credit to the
creators of the Arduino SDK of which I modified and used. Where/whom would
you like me to give credit to?

PS. will hopefully upload my modified version today.

Cheers,
Matt

sitanshu1, 2018-02-05:
@MatMcT 
Hello,
     I am not sure about the credits to give to. But can you please upload the code if it is done? I am having issues with the anticlockwise rotations of the pro motor as hex code cannot be given in negative integers.

Thank you
Sitanshu

MatMcT, 2018-02-06:
Hello everyone,

Apologies for the late reply. I've created a repo with my adapted library:
https://bitbucket.org/mattmct/dxl-pro-teensy

Simply ADD the contents of the "libraries" folder into the same location as your other Arduino libraries are located.

@sitanshu1 I'm not sure what application you're using the motor for but if you look at the DXL.cpp file, you can see how I wrote my 'DXLsetPos' function. You can actually set positive and negative angles. For me, all I needed was to set absolute angles and this suited me fine. You may be able to adapt my code to suit you.

The control tables helped me quite a bit when understanding the original code:
http://support.robotis.com/en/product/actuator/dynamixel_pro/dynamixelpro/control_table.htm#Actuator_Address_11