ROBOTIS-GIT/OpenCR #39: Cannot compile OpenCR codes

Affonso-Gui, 2017-04-21:
When trying to compile any code under OpenCR Board option from Arduino IDE, I get the following error:
```
Arduino: 1.8.1 (Linux), TD: 1.35, Board: "OpenCR Board, OpenCR Bootloader"

fork/exec /home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++: no such file or directory
Error compiling for board OpenCR Board.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

When enabled 'Show verbose output during compilation' got this one:
```
Arduino: 1.8.1 (Linux), TD: 1.35, Board: "OpenCR Board, OpenCR Bootloader"

/home/robotis/arduino-1.8.1/arduino-builder -dump-prefs -logger=machine -hardware /home/robotis/arduino-1.8.1/hardware -hardware /home/robotis/.arduino15/packages -tools /home/robotis/arduino-1.8.1/tools-builder -tools /home/robotis/arduino-1.8.1/hardware/tools/avr -tools /home/robotis/.arduino15/packages -built-in-libraries /home/robotis/arduino-1.8.1/libraries -libraries /home/robotis/Arduino/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10801 -build-path /tmp/arduino_build_293874 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_tools.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_gcc.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -verbose /tmp/untitled1520870712.tmp/sketch_apr21a/sketch_apr21a.ino
/home/robotis/arduino-1.8.1/arduino-builder -compile -logger=machine -hardware /home/robotis/arduino-1.8.1/hardware -hardware /home/robotis/.arduino15/packages -tools /home/robotis/arduino-1.8.1/tools-builder -tools /home/robotis/arduino-1.8.1/hardware/tools/avr -tools /home/robotis/.arduino15/packages -built-in-libraries /home/robotis/arduino-1.8.1/libraries -libraries /home/robotis/Arduino/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10801 -build-path /tmp/arduino_build_293874 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_tools.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_gcc.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -verbose /tmp/untitled1520870712.tmp/sketch_apr21a/sketch_apr21a.ino
Using board 'OpenCR' from platform in folder: /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11
Using core 'arduino' from platform in folder: /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11
Detecting libraries used...
"/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10801 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr/include" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/driver" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/usb_cdc" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/cores/arduino" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR" "/tmp/arduino_build_293874/sketch/sketch_apr21a.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10801 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr/include" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/driver" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/usb_cdc" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/cores/arduino" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR" "/tmp/arduino_build_293874/sketch/sketch_apr21a.ino.cpp" -o "/tmp/arduino_build_293874/preproc/ctags_target_for_gcc_minus_e.cpp"
fork/exec /home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++: no such file or directory
Error compiling for board OpenCR Board.
```

Is there some step I am missing? I am using OpenCR version 1.0.11 and basically following instructions on [Turtlebot3 Wiki](http://turtlebot3.robotis.com/en/latest/opencr_software.html).

ROBOTIS-Leon, 2017-04-23:
Hi, 

Can you try this again?

``` bash
$ sudo apt-get install libncurses5-dev:i386
```

Affonso-Gui, 2017-04-27:
Was able to compile successfully after running the command!

As I had some custom messages defined in Arduino/libraries/ros_lib I had to update it too, but everything ran fine after it. 
(At first I got the following error message:
```
 Arduino: 1.8.1 (Linux), TD: 1.35, Board: "OpenCR Board, OpenCR Bootloader"

/home/robotis/arduino-1.8.1/arduino-builder -dump-prefs -logger=machine -hardware /home/robotis/arduino-1.8.1/hardware -hardware /home/robotis/.arduino15/packages -tools /home/robotis/arduino-1.8.1/tools-builder -tools /home/robotis/arduino-1.8.1/hardware/tools/avr -tools /home/robotis/.arduino15/packages -built-in-libraries /home/robotis/arduino-1.8.1/libraries -libraries /home/robotis/Arduino/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10801 -build-path /tmp/arduino_build_335622 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_gcc.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_tools.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -verbose /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/libraries/turtlebot3/turtlebot3_waffle/turtlebot3_core/turtlebot3_core.ino
/home/robotis/arduino-1.8.1/arduino-builder -compile -logger=machine -hardware /home/robotis/arduino-1.8.1/hardware -hardware /home/robotis/.arduino15/packages -tools /home/robotis/arduino-1.8.1/tools-builder -tools /home/robotis/arduino-1.8.1/hardware/tools/avr -tools /home/robotis/.arduino15/packages -built-in-libraries /home/robotis/arduino-1.8.1/libraries -libraries /home/robotis/Arduino/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10801 -build-path /tmp/arduino_build_335622 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_gcc.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_tools.path=/home/robotis/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -verbose /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/libraries/turtlebot3/turtlebot3_waffle/turtlebot3_core/turtlebot3_core.ino
Using board 'OpenCR' from platform in folder: /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11
Using core 'arduino' from platform in folder: /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11
Detecting libraries used...
"/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10801 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr/include" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/driver" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/usb_cdc" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/cores/arduino" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR" "/tmp/arduino_build_335622/sketch/turtlebot3_core.ino.cpp" -o "/dev/null"
"/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10801 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/bsp/opencr/include" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/driver" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/hw/usb_cdc" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR/" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/cores/arduino" "-I/home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/variants/OpenCR" "-I/home/robotis/Arduino/libraries/ros_lib" "/tmp/arduino_build_335622/sketch/turtlebot3_core.ino.cpp" -o "/dev/null"
"/home/robotis/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 [... cut ...]
 Not used: /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/libraries/turtlebot3_ros_lib
In file included from /home/robotis/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.11/libraries/turtlebot3/turtlebot3_waffle/turtlebot3_core/turtlebot3_core.ino:19:0:
/tmp/arduino_build_335622/sketch/turtlebot3_core_config.h:35:41: fatal error: turtlebot3_msgs/SensorState.h: No such file or directory
compilation terminated.
Using library ros_lib in folder: /home/robotis/Arduino/libraries/ros_lib (legacy)
exit status 1
Error compiling for board OpenCR Board.
```

Thank you very much

ROBOTIS-Leon, 2017-04-27:
@Affonso-Gui 

Thank you for information

closed 2017-04-27 by ROBOTIS-Leon

Omkar-Jadhav, 2021-05-18:
?Hello, @Affonso-Gui ,
I am having the same issue after using a custom message. Can you share how did you update Arduino/libraries/ros_lib?

Affonso-Gui, 2021-05-19:
@Omkar-Jadhav It has been many years since I last touched an arduino and I don't quite remember what I did at that time, but since I don't recall it being anything particularly complicated I'm guessing I just kind of re-generated the messages and header files?

Maybe something like this: http://wiki.ros.org/rosserial_arduino/Tutorials/Adding%20Custom%20Messages