ROBOTIS-GIT/OpenCR #50: CAN driver

martinichka, 2017-10-25:
Dear ROBOTIS,

For a project we want to use motors via CAN-open. The OpenCR board has CAN, but we can't find any source code supporting it.

We are planning to write the CAN driver and Arduino support for it. We will fork https://github.com/ROBOTIS-GIT/OpenCR/ and add Arduino support for the STM driver /variants/OpenCR/lib/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_can.c.

Therefore I have just a few questions:

1. First of all: Isn't there already already Arduino-STM32F7-CAN support in a hidden place where I couldn't find?
2. Is our plan the way to go?
3. Do you see any drawbacks?
4. What is your development environment? In some folder I saw some Eclipse project files, but what do you use for Arduino Board package development?

Best regards,
Marten

closed 2017-12-19 by martinichka

chcbaram, 2018-01-04:
1. First of all: Isn't there already already Arduino-STM32F7-CAN support in a hidden place where I couldn't find?
  -> we are working on the can driver so you can use the can APIs In a short time.

2. Is our plan the way to go?
  -> Yes, you can use stm32F7 HAL driver in OpenCR folder but you have to modify OpenCR source code.

3. Do you see any drawbacks?
   -> I think if you add your own can driver, it will not be compatible in the future.  So after we finish the can driver for OpenCR, you can check if you can use the driver without modification.

4. What is your development environment? In some folder I saw some Eclipse project files, but what do you use for Arduino Board package development?
   -> it's just built with gcc. The Eclipse project files was just to edit the source code not for build.

amburkoff, 2018-11-12:
When will the source code driver CAN for openсr?