ROBOTIS-GIT/OpenCR #54: OpenCR1.0 turtlebot3_motor_setup

ludusrusso, 2017-12-06:
Hi,
I've recently bought a Turtlebot3 but one motor does not communicate. 
I've try using the `turtlebot3_motor_setup` but it seems not able to find any motor. Both with the working motor attached or not. 

Any solution? Unfortunatly I've not a USB2Dynamixel and it's complicate to buy one!

ROBOTIS-Leon, 2017-12-07:
@ludusrusso 

Hi, 

here is the source that makes the OpenCR run as USB2Dynamixel.
Just download this source (arduino code) to the OpenCR board.

https://github.com/ROBOTIS-GIT/OpenCR/tree/develop/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc

Questions are welcomed

ludusrusso, 2017-12-07:
Hi!
Thanks for your reply... 
I'm still not able to communicate with the robot. Since I've linux, I've tried to use this package in ROS, but still no results with both the working robot and the other one...

https://github.com/ROBOTIS-GIT/dynamixel-workbench

I got this error:

```
SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyACM0
 * /ping: False
 * /ping_id: 2
 * /protocol_version: 2.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
    single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

auto-starting new master
process[master]: started with pid [1726]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3adb1012-db58-11e7-acb8-080027f27167
process[rosout-1]: started with pid [1743]
started core service [/rosout]
process[single_dynamixel_monitor-2]: started with pid [1750]
process[single_dynamixel_controller-3]: started with pid [1754]
[ INFO] [1512655765.647136603]: Succeeded to open the port(/dev/ttyACM0)!
[ INFO] [1512655765.710935057]: Succeeded to change the baudrate(1000000)!
[ INFO] [1512655765.711474518]: Scan Dynamixel(ID: 1~253) Using Protocol 2.0

[ INFO] [1512655765.711635683]: ...wait for seconds
[ WARN] [1512655770.299463407]: Please Check USB Port authorization and
[ WARN] [1512655770.299534010]: Baudrate [ex : 57600, 115200, 1000000, 3000000]
[ERROR] [1512655770.299984694]: ...Failed to find dynamixel!
================================================================================REQUIRED process [single_dynamixel_monitor-2] has died!
process has died [pid 1750, exit code -11, cmd /home/hbr/dyn_ws/devel/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/hbr/.ros/log/3adb1012-db58-11e7-acb8-080027f27167/single_dynamixel_monitor-2.log].
log file: /home/hbr/.ros/log/3adb1012-db58-11e7-acb8-080027f27167/single_dynamixel_monitor-2*.log
Initiating shutdown!
================================================================================
[single_dynamixel_controller-3] killing on exit
[single_dynamixel_monitor-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

any idea?

ludusrusso, 2017-12-07:
Update:

Tested with Windows and Dynamixel Firmware Recovery. Same issue... The robot communicate but does not find motors!

ROBOTIS-Leon, 2017-12-07:
@ludusrusso

hi, again.

change the baudrate of the motor from 57600(default) into 1000000 after firmware recovery(this was done).

or

change the access baudrate in the script of single_manager.launch into 57600 ( which might be set as default in the Dynamixel)

ludusrusso, 2017-12-07:
Hi!
I’ve tried both! It is possible that the script on the opencr communicate with the dynamixel port with 4 pins? Become i’m using the one with 3 pins!

ROBOTIS-Leon, 2017-12-07:
@ludusrusso

hi, again

4pin and 3pin circuit are connected, and you dont need to choose either of them.
Once it tx the packet, it goes to both

ROBOTIS-Leon, 2017-12-07:
@ludusrusso

okay, dont worry

let me check your environment again.

you're using 

couple of xl430-350 with newest firmware updated(41)
and those are set in:
baudrate : 1Mbps
ID: 1 and 2, each, and connected with 3pin cable

youve just changed the firmware of the opencr board from the turtlebot core to usb-to-dxl what I mentioned previously.

and now you downloaded the ROS dynamixel-workbench and it couldn't detect both anyhow

right?

did you set the IDs of those to 1 and 2? first and foremost

ROBOTIS-Leon, 2017-12-07:
sorry, xl430-250

ludusrusso, 2017-12-07:
It’s not a problem! just trying to understand!!
Yes to all, with one more thing: i’ve Also downloaded the roboplus tool and using the dynamixel wizard tried to reset/communicate with the motors.. in both cases, the tool with opencr and usb-to-dxl is not able to find any robot! This is why I think the problem is somewhere in the usb-to-dxl or hardware :)

ludusrusso, 2017-12-07:
Just to be clear: the main problem is that i’m not able to set the ID to one motor. The motor with ID 2 works, the other one does not communicate with turtlebot_core and i’m trying to reset it!

But a part from the turtlebot_core program, I don’t find others ways to communicate with both motors.. I’ve tried several scripts in the examples of dynamixel_sdk 2.0 on arduino ide and using the usb-to-dxl

ROBOTIS-Leon, 2017-12-07:
@ludusrusso

kay, I will find out tomorrow morning

stay tuned

ludusrusso, 2017-12-08:
Hi,
is there a particular configuration I've to set with the ust_to_dxl code? I'm quite sure I'm missing some stupid stuff.

Here is my configuratio:

 - OpenCR with usb_to_dxl
 - USB connected to the micro-USB of the OpenCR
 - Motor connected to the 3PIN (the middle one) port

After fushing the usb_to_dxl code I cat this stuff (that in my understand means "ok")

```
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /var/folders/l_/vgnmwcw958z27mrd7b45tflc0000gn/T/arduino_build_540092/usb2dyn.ino.bin 
file size : 107 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 1.041000 sec
flash_write : 0 : 0.563000 sec 
CRC OK A9EED6 A9EED6 0.003000 sec
[OK] Download 
jump_to_fw 
```

The OpenCR is also powered by the turtlebot3 battery. When I power on it, the motor led blinks one time! 

But when I open the Dynamixel Wizard program and try to do something (not important what). Only the led User2 blinks. If I correctly understand, it means that the device does not respond to the PC...

ludusrusso, 2017-12-10:
@ROBOTIS-Leon any news?

ROBOTIS-Will, 2017-12-11:
@ludusrusso Hello,   
If you are using Dynamixel Wizard, please switch the software to [R+ Manager 2.0](http://en.robotis.com/BlueAD/board.php?bbs_id=downloads&mode=view&bbs_no=1152559&page=1&key=&keyword=&sort=&scate=) as Dynamixel X series are not supported in Dynamixel Wizard.

ludusrusso, 2017-12-11:
@ROBOTIS-Will thanks!
But again, same issues! No motor is detected with the R+ Manager 2.0!
Moreover, in firmware recovery, I cannot selected my model, that is XL-430 W250-T

[image]

ROBOTIS-Will, 2017-12-12:
@ludusrusso 
Please select XL430-W250 from the Home screen and proceed the firmware recovery as shown below.  
[image]  
If it doesn't detect your XL430-W250, then please try again with other DXL_BAUD from usb_to_dxl.ino file, such as 57600.

ludusrusso, 2017-12-12:
Ok thanks!
Now I have some improvements, but still the same issue!
When I try to do system "auto search", no motor is discovered but the RX led blinks! So now the motor is sending something..

However, when I try to do the recovery procedure, nothing happens, also the RX led does not blinks...

I've tried different configuration of Serial1 baud and Serial3 baud in the usb_to_dxl..

There is something other I can do?

ludusrusso, 2017-12-12:
Update! Strange fact: I noticed that te User1 led blinks also if no motor is connected. By reading the usb_to_dxl program, it should blinks only if it receives data from Serial3... My question is: is Serial3 the correct dynamixel port? Is there something other attached to it?

ROBOTIS-Will, 2017-12-13:
@ludusrusso 
First of all, please confirm that your R+Manager 2.0 is up to date and select XL430 from the home screen.  
If possible, please upload the usb_to_dxl.ino file that you used, and any recorded video for your procedure would greatly helpful.  
Thank you.

ludusrusso, 2017-12-13:
Hi!
Yes, I've updated R+Manager 2.0 and selected the XL430...

I'm using the version of dxl_to_usb.ino linked above, I'll copy it here:

```
#include <DynamixelSDK.h>
#include <stdarg.h>

#define DXL_USB_VER           20170915

#define CMD_PORT              Serial      // USB
#define DBG_PORT              Serial2     // UART1
#define DXL_PORT              Serial3
#define DXL_BAUD              1000000

#define DXL_LED_RX            BDPIN_LED_USER_1
#define DXL_LED_TX            BDPIN_LED_USER_2

#define DXL_POWER_DISABLE()   digitalWrite(BDPIN_DXL_PWR_EN, LOW);
#define DXL_POWER_ENABLE()    digitalWrite(BDPIN_DXL_PWR_EN, HIGH);

#define DXL_TX_BUFFER_LENGTH  1024

uint8_t tx_buffer[DXL_TX_BUFFER_LENGTH];

static int rx_led_count = 0;
static int tx_led_count = 0;

static int rx_led_update_time;
static int tx_led_update_time;

static uint32_t update_time[8];

static uint32_t rx_data_cnt = 0;
static uint32_t tx_data_cnt = 0;

static uint32_t rx_bandwidth = 0;
static uint32_t tx_bandwidth = 0;

uint32_t usb_baud;

void setup()
{
  CMD_PORT.begin(115200);
  DBG_PORT.begin(57600);
  DXL_PORT.begin(DXL_BAUD);

  pinMode( BDPIN_DXL_PWR_EN, OUTPUT );
  pinMode( DXL_LED_RX, OUTPUT );
  pinMode( DXL_LED_TX, OUTPUT );

  digitalWrite(DXL_LED_TX, HIGH);
  digitalWrite(DXL_LED_RX, HIGH);

  drv_dxl_tx_enable(FALSE);

  DXL_POWER_ENABLE();
}

void loop()
{
  int length;
  int i;

  update_dxl();
  update_led();

  if( CMD_PORT.getBaudRate() != DXL_PORT.getBaudRate() )
  {
    DXL_PORT.begin(CMD_PORT.getBaudRate());
  }

  if( (millis()-update_time[1]) > 1000 )
  {
    update_time[1] = millis();

    tx_bandwidth = tx_data_cnt;
    rx_bandwidth = rx_data_cnt;

    tx_data_cnt = 0;
    rx_data_cnt = 0;
  }
}

void update_dxl()
{
  int length;
  int i;

  //-- USB -> DXL
  length = CMD_PORT.available();
  if( length > 0 )
  {
    drv_dxl_tx_enable(TRUE);
    for(i=0; i<length; i++ )
    {
      DXL_PORT.write(CMD_PORT.read());
    }
    drv_dxl_tx_enable(FALSE);

    tx_led_count = 3;

    tx_data_cnt += length;
  }

  //-- DXL -> USB
  length = DXL_PORT.available();
  if( length > 0 )
  {
    if( length > DXL_TX_BUFFER_LENGTH )
    {
      length = DXL_TX_BUFFER_LENGTH;
    }
    for(i=0; i<length; i++ )
    {
      tx_buffer[i] = DXL_PORT.read();
    }
    CMD_PORT.write(tx_buffer, length);

    rx_led_count = 3;
    rx_data_cnt += length;
  }
}

void update_led()
{
  if( (millis()-tx_led_update_time) > 50 )
  {
    tx_led_update_time = millis();

    if( tx_led_count )
    {
      digitalWrite(DXL_LED_TX, !digitalRead(DXL_LED_TX));
      tx_led_count--;
    }
    else
    {
      digitalWrite(DXL_LED_TX, HIGH);
    }
  }

  if( (millis()-rx_led_update_time) > 50 )
  {
    rx_led_update_time = millis();

    if( rx_led_count )
    {
      digitalWrite(DXL_LED_RX, !digitalRead(DXL_LED_RX));
      rx_led_count--;
    }
    else
    {
      digitalWrite(DXL_LED_RX, HIGH);
    }
  }
}
```

I've only changed the DXL_BAUD to the various baud rate supported by the dxl... 9600 15200 etc.

I'll done a video tomorrow

ludusrusso, 2018-01-02:
Hi!
I've tested the motor with U2D2 and it does not responds. 
Strange fact: the RX led blinks together with the TX led even if the motor is not connected to the power. I'm quite sure it is an electronic demage! I'll close the issue!

closed 2018-01-02 by ludusrusso

Ash-S, 2020-10-27:
Hi @ROBOTIS-Will, I am facing the similiar issue while interfacing OpenCR with RaspberryPi3. I tried to use openCR as U2D2 while uploading 'usb_to_dxl' scatch from Arduino. But before uploading when I compiled it, it failed. Please auggest. Thanks.

ROBOTIS-Will, 2020-10-27:
Hi @Ash-S 
The OpenCR does not support Arduino IDE on ARM based CPU such as Raspberry Pi and Jetson Nano.
If you are trying to upload your sketch using Arduino IDE, you should upload it from your PC.
Thanks.

Ash-S, 2020-10-27:
Hi @ROBOTIS-Will , thanks for your prompt reply. 
well, I am trying to upload it through PC not from SBC. The error message details are here.... 
Please help.

```
Detecting libraries used...
"C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10809 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/bsp/opencr" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/bsp/opencr/include" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/hw" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/hw/driver" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/hw/usb_cdc" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\cores\arduino" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR" "C:\Users\ASHCOM\AppData\Local\Temp\arduino_build_9377\sketch\usb_to_dxl.ino.cpp" -o "nul"
"C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10809 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/bsp/opencr" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/bsp/opencr/include" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/hw" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/hw/driver" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/hw/usb_cdc" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR/" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\cores\arduino" "-IC:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\variants\OpenCR" "C:\Users\ASHCOM\AppData\Local\Temp\arduino_build_9377\sketch\usb_to_dxl.ino.cpp" -o "C:\Users\ASHCOM\AppData\Local\Temp\arduino_build_9377\preproc\ctags_target_for_gcc_minus_e.cpp"
In file included from C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\OpenCR\hardware\OpenCR\1.4.16\cores\arduino/Arduino.h:23:0,

                 from C:\Users\ASHCOM\AppData\Local\Temp\arduino_build_9377\sketch\usb_to_dxl.ino.cpp:1:

c:\programdata\matlab\supportpackages\r2018a\3p.instrset\arduinoide.instrset\idepkgs\packages\opencr\tools\opencr_gcc\5.4.0-2016q2\lib\gcc\arm-none-eabi\5.4.1\include\stdint.h:9:26: fatal error: stdint.h: No such file or directory

compilation terminated.

exit status 1
```
Error compiling for board OpenCR Board.

ROBOTIS-Will, 2020-10-27:
If you are compiling the sketch from other software than Arduino IDE, I cannot guarantee its success.
Please try with the Arduino IDE as written in the [eManual](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-on-windows).
Thanks.

Ash-S, 2020-10-27:
@ROBOTIS-Will I am only compiling it with Arduino IDE version 1.8.9 on windows 10.

ROBOTIS-Will, 2020-10-27:
According to your error, it looks like your compiler cannot find the header file.
Make sure that you have appended the gcc compiler to the windows environment path.
The reason I thought you are using other software was because the path of the library was not the default, but included in the matlab directory.
Mine looks like below
```
Detecting libraries used...
"C:\\Users\\WillSon\\AppData\\Local\\Arduino15\\packages\\OpenCR\\tools\\opencr_gcc\\5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LE...
```

Ash-S, 2020-10-28:
Hi @ROBOTIS-Will thanks for your suggestion but 'gcc' is already appended in my windows env path. Also i didn't understand why this openCR package get installed in MATLAB support package. Do I need to change the install path some where?  If yes, then where I have to dig into it, please suggest.

ROBOTIS-Will, 2020-10-28:
I'd suggest to completely remove the Arduino IDE and reinstall with default settings.
Then, install OpenCR board manager using Arduino IDE.
Thank you.

Ash-S, 2020-11-02:
Hi @ROBOTIS-Will  thanks. Seems that was the only solution when nothing works. :) 
I re-installed IDE and compiled/uploaded the sketch to openCR well. 
later, I followed the emanual of openCR which is connected to RaspberryPi. I can access the port '/dev/ttyACM0'. I can run the `"rosrun dynamixel_workbench_controller dynamixel_workbench_controller /dev/ttyACM0"` well and scan the 4 dynamixels connected to the openCR. I updated the /dev/ttyACM0 to the launch and others where required. But I failed at 'roslaunch',
`"roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch"`. gives me this error:

[image]

ROBOTIS-Will, 2020-11-02:
Hi @Ash-S 
As your inquiry is completely out of the original issue and repository, please open an new issue ticket on DYNAMIXEL Workbench repository.
Thank you.

Ash-S, 2020-11-03:
Okay @ROBOTIS-Will . I am deleting the query from this repository. Thanks.