ROBOTIS-GIT/OpenCR #57: How do I access to register to set PWM frequency and resolution with OpenCR?

EdXian, 2018-01-13:
Hi ROBOTIS team,
I want to do a drone project with stm32 OpenCR Board . Now I have to set PWM Pins to 400 Hz and create some couters for my radio controller but I can't found the low layer function for me to set the peripheral in chip. 
My question is that "How do I access to register and set the peripheral "?

chcbaram, 2018-01-15:
The pwm driver file is blow link.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/hw/driver/drv_pwm.c

You can modify the driver source for PWM. 
But you can try the functions for PWM, there are two low level functions for PWM and you can change the frequency for PWM.

void drv_pwm_set_freq(uint32_t ulPin, uint32_t freq_data)
void drv_pwm_setup(uint32_t ulPin)

EdXian, 2018-01-17:
HI @chcbaram ,
Thank you for your quick reply!
I have some problem when trying to set  pwm frequency below 10Khz.Everything is fine if I set PWM frequency greater than 10Khz.

400Hz:
```
drv_pwm_set_freq(pwm_pins[0],400);  //set 400hz
drv_pwm_set_duty(pwm_pins[0],10,500); //set duty 50%
```
[image]

10KHz:
```
drv_pwm_set_freq(pwm_pins[0],10000);  //set 10khz
drv_pwm_set_duty(pwm_pins[0],10,500); //set duty 50%
```
[image]

Thanks in Advance!

OpusK, 2018-01-17:
Hi,

According to [Reference Manual](http://www.st.com/content/ccc/resource/technical/document/reference_manual/c5/cf/ef/52/c0/f1/4b/fa/DM00124865.pdf/files/DM00124865.pdf/jcr:content/translations/en.DM00124865.pdf) of MCU used in OpenCR, Auto-reload register is 16bit except TIM2,3,4,5.

I do not know the pin connected to your pwm_pins [0]. 
But if you divide the 168Mhz clock by 400hz, there will be an overflow for 16bits, which may cause a value of 13.7474kHz depending on the timer.

First of all, please check the number of the timer (16bit or 32bit) assigned to the pin you are going to use.

This information can be found in [variant.cpp.](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/variant.cpp)

EdXian, 2018-01-18:
Hi @OpusK ,
I use 'Serial.println(SystemCoreClock);'  and get 216000000 . I assume that the core_clock is 216Mhz and try to change the prescalar to non-zero value  to prevent the 16-bit register from overflow problem([here](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/hw/driver/drv_pwm.c#L131))
But I still cant get the desired PWM output.
Thanks

OpusK, 2018-01-19:
Oops.. Sorry. 
I gave you the wrong information.
TIMER2,5 only ARR register is 32bit.

I actually tested it with the code below and it worked at 400hz.
```c
#define PWM_PIN 6

void setup() {
  drv_pwm_set_freq(PWM_PIN, 400);
  drv_pwm_setup(PWM_PIN);  
  drv_pwm_set_duty(PWM_PIN, 10, 512);
}

void loop() {
}
```

And, as you say, you can adjust the value that overflows through the prescaler. 
Can you see your code? Did you calculate and change the period value?

OpusK, 2018-01-19:
Oops.. Sorry. 
I gave you the wrong information.
TIMER2,5 only ARR register is 32bit.

I actually tested it with the code below and it worked at 400hz.
```c
#define PWM_PIN 6

void setup() {
  drv_pwm_set_freq(PWM_PIN, 400);
  drv_pwm_setup(PWM_PIN);  
  drv_pwm_set_duty(PWM_PIN, 10, 512);
}

void loop() {
}
```

> try to change the prescalar to non-zero value to prevent the 16-bit register from overflow problem(here)

As you say, you can use the prescaler to adjust the overflow value.
Can I see your code? Did you calculate and change the period value? (please refer to [here](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/hw/driver/drv_timer.c#L257))

EdXian, 2018-01-19:
Hi @OpusK ,
Thank you for your quick reply.
The code you provide works for me . In fact , I want to use PWM to control four ESCs which can control the speed of brushless motor, so I need at least 4 PWM pins . After the test , I found only D6 could work and  D3 D5 D9 D10 D11 couldn't work. 
According to  [variant.cpp](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/variant.cpp#L46-L52) and [drv_timer.c](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/hw/driver/drv_timer.c#L257) I found the coresponding timers and channels for each PWM pins as following list:
TIM3 | TIM1 | TIM2  | TIM9  | TIM11  | TIM12 
D3 | D5 | D6  | D9  | D10  | D11 
ch1 | ch1 | ch3 | ch2 | ch1 | ch2

You can see my code as following:
```
int pwm_pins[5] = { 3, 5, 9, 10, 11 };

void setup() {
  int j =0;
  for(j=0;j<5;j++){
    
    //drv_timer_set_period(uint8_t channel, uint32_t period_data)  // How do I set this function?
    //  (216000000 / prescaler+1 )/(400Hz) /()  =  period.data+1  //Is it correct ?
    
    drv_pwm_set_freq(pwm_pins[j], 400);  //set 400 HZ PWM
    drv_pwm_setup(pwm_pins[j]);  
     drv_pwm_set_duty(pwm_pins[j], 12,3075);  // duty 75% 
  }
}

void loop() {
 
  int i;
  for(i =0;i<5;i++){
     drv_pwm_set_duty(pwm_pins[i], 12,4096*0.5-1); //duty 50%
  }
  delay(1000);
   for(i =0;i<5;i++){
   
  drv_pwm_set_duty(pwm_pins[i], 12,4096*0.8-1);  //duty 80%
  }
  delay(1000);
}

```
Can you help me to set the functiondrv_timer_set_period( , )?
Thanks in advance!!

EdXian, 2018-01-19:
Hi,
Thanks again!
According to the fomula: (core_clock)/[ (period+1)* (prescaler+1)] = frequency.
I finally calculate the value of `prescaler` & `period` and customize the drv_pwm function as below. It  works fine!
 
```
void pwm_setup(uint32_t ulPin )
{
  TIM_HandleTypeDef  *pTIM;
  TIM_OC_InitTypeDef *pOC;
  uint32_t tim_ch;
  uint32_t uwPeriodValue;

  if( ulPin >= PINS_COUNT )     return;
  if( pwm_init[ulPin] == true ) return;

  pTIM   = g_Pin2PortMapArray[ulPin].TIMx;
  tim_ch = g_Pin2PortMapArray[ulPin].timerChannel;
  uwPeriodValue = (uint32_t) (((SystemCoreClock)  / 400) - 1);

  if( pTIM == &hTIM3 )
  {
    pOC = &hOC3;
    pTIM->Instance = TIM3;
  }
  else if( pTIM == &hTIM1 )
  {
    uwPeriodValue = (uint32_t) (((SystemCoreClock)  / 400) - 1);
    pOC = &hOC1;
    pTIM->Instance = TIM1;
  }
  else if( pTIM == &hTIM2 )
  {
    pOC = &hOC2;
    pTIM->Instance = TIM2;
  }
  else if( pTIM == &hTIM9 )
  {
    uwPeriodValue = (uint32_t) (((SystemCoreClock)  /400) - 1);
    pOC = &hOC9;
    pTIM->Instance = TIM9;
  }
  else if( pTIM == &hTIM11 )
  {
    uwPeriodValue = (uint32_t) (((SystemCoreClock)  / 400) - 1); //set 400HZ
    pOC = &hOC11;
    pTIM->Instance = TIM11;
  }
  else if( pTIM == &hTIM12 )
  {
    pOC = &hOC12;
    pTIM->Instance = TIM12;
  }
  else
  {
    return;
  }

  pTIM->Init.Prescaler         = 215;  //216  -1
  pTIM->Init.Period            = 2499; //2500 -1
  pTIM->Init.ClockDivision     = TIM_CLOCKDIVISION_DIV1;
  pTIM->Init.CounterMode       = TIM_COUNTERMODE_UP;
  pTIM->Init.RepetitionCounter = 0;
  HAL_TIM_PWM_Init(pTIM);

  memset(pOC, 0, sizeof(TIM_OC_InitTypeDef));

  pOC->OCMode       = TIM_OCMODE_PWM1;
  pOC->OCPolarity   = TIM_OCPOLARITY_HIGH;
  pOC->OCFastMode   = TIM_OCFAST_DISABLE;
  pOC->OCNPolarity  = TIM_OCNPOLARITY_HIGH;
  pOC->OCNIdleState = TIM_OCNIDLESTATE_RESET;
  pOC->OCIdleState  = TIM_OCIDLESTATE_RESET;

  pOC->Pulse = 0;
  HAL_TIM_PWM_ConfigChannel(pTIM, pOC, tim_ch);
  HAL_TIM_PWM_Start(pTIM, tim_ch);
  pwm_init[ulPin] = true;
}
```

OpusK, 2018-01-19:
Since each value is a constant, the desired frequency may not be obtained depending on the timer characteristics and other settings.

If you have problems later, try creating a function that can manage prescaler values, as I had previously answered. 

I wish you well.
Thanks :)!

closed 2018-01-19 by OpusK

OpusK, 2018-02-01:
Hi, @EdXian 

We have modified the code in relation to this issue.
It will be included in the next release and is now available in [the develop branch](https://github.com/ROBOTIS-GIT/OpenCR/tree/ba48db8b2cb22c3da75f24f94045d19c1c73f7a9).

If you need it, you can download and use it.
Thanks :)

EdXian, 2018-02-02:
Hi @OpusK ,
Great ! Thank you for your help!