ROBOTIS-GIT/OpenCR #58: OpenCR1.0 Bord Dynamixel port max current

newrobofreak, 2018-01-14:
Hi,
I currently started to work with the Dynamixel X-Series and the OpenCR board.
For my project I would like to daisy chain Dynamixel servos (XL430-W250 or XM430-W210) and connect them to the OpenCR board.
My question is how much current can I draw from the board. For Example, would two XL430-W250 at full load work? Where is the limit from which I need external power supply?
Could not find any information about the max supply current from the board in the manual and would be glad for any help.
:)

OpusK, 2018-01-15:
OpenCR has 10A Fuse.
Therefore, it will not work if a current of around 10A or more flows.

Therefore, the XL430-W250 has a specification of 1.5 Nm @ 12.0V, 1.4A, so there is no theoretical problem with two Full Loads. (For more information, please refer to [the manual](http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#specifications).)

newrobofreak, 2018-01-15:
Thanks a lot. 10A was the value, I was looking for.
So, that means I could operate **6x XL430-W250** [1.5Ax6=9A<10A] or **3x XM430-W210** with 3.7N.m (at 14.8V, 2.7A) [2.7Ax3=8.1A<10A] through just **one OpenCR1.0 port** by daisy chaining them? And because DXM_PWR is the same connector there should be no difference between TTL and RS-485, right?

OpusK, 2018-01-16:
Yes, you are right.

The same power source (DXL_PWR in the schematic) is used for the TTL and RS-485 connectors, so you can do so.
The daisy chaining method is no problem in theory, and it is actually used in many cases. (Except for the circuit and wire resistance and ambient environment problems, this is normal)

newrobofreak, 2018-01-16:
Thanks a lot for answering that fast and competent. Great support here. I Appreciate it.

closed 2018-01-16 by newrobofreak