ROBOTIS-GIT/OpenCR #62: DYNAMIXEL XM430-W210-R with OpenCR

flg-vs, 2018-01-23:
Hi,
I am trying to setup the OpenCR Board to control four DYNAMIXEL XM430-W210-R by daisychaining them. However unfortunately, none of these motors seem to work. 
I already tried multiple examples from OpenCR -> DynamixelWorkbench such as t_Find_Dynamixel. However none of these where able to find any Motors, even though I adapted them to use all available baudrates and ranges from 0 to 254 on the scans. (both when all four and also when only one motor was attached). 
I also tried to use the usb_to_dxl Example to use the OpenCR board as a Dynamixel2Usb adapter as suggested in [this issue](https://github.com/ROBOTIS-GIT/OpenCR/issues/54#issuecomment-349867492), but with Windows7 the board was not correctly identified so I could not test the motors with RoboplusManager 2.

Other than the issue with the motors, the board seems to be running completely fine. With the default firmware, reading IMU data, etc... works perfectly.

I hope that you could help me find the problem with the motors. 

Thank you very much and kind regards!

OpusK, 2018-01-24:
Hi, @flg-vs 

We have a WIN7 user in our company and have tested it.
As a result, the motor was searched without any problem.

When searching with R + Manager, have you selected [ALL] and the entire baud rate?
If not, please try it again. (Only one motor)

And did you turn on the power for the motor in OpenCR? If so, how many V, A?

flg-vs, 2018-01-29:
Hi @OpusK Thank you for the feedback. 
The problem with the windows computer was that the STM32 driver wasn't correctly installed. Once I fixed this the board was correctly recognised. 
The issue with the motors turned out to be a hardware problem as I made a mistake when I daisychained them. They are working fine now.

Thank you for your help and kind regards

closed 2018-01-29 by flg-vs