ROBOTIS-GIT/OpenCR #63: RTOS library cannot be included by other source files?

EdXian, 2018-01-24:
Hi team,
I tried to include RTOS.h to my source file as following . However the compiler return error : 
```
.arduino15/packages/OpenCR/hardware/OpenCR/1.0.15/libraries/RTOS/src/RTOS.h:1:8: error: expected identifier or '(' before string constant
 extern "C"
        ^
exit status 1
Error compiling for board OpenCR Board.
```
task.h
```
#ifndef _STABLE_H
#define _STABLE_H
#include "RTOS.h"
#ifdef __cplusplus
 extern "C" {
#endif
 void task(void const *argument);

#ifdef __cplusplus
}
#endif
#endif
``` 
task.c
```
#include "stable.h"
void task(void const *argument)
{
  (void) argument;
  while(1){
   digitalWrite(13,!digitalRead(13));
    osDelay(1000);
  }
}
```
It works fine if I modify the `libraries/RTOS/src/RTOS.h`  like following:
```
#ifdef __cplusplus
 extern "C" {
#endif
 #include "cmsis_os.h"
#ifdef __cplusplus
}
#endif
```
Any idea?

OpusK, 2018-01-24:
Hi, @EdXian 

RTOS.h was designed as a header file included in cpp.
Therefore, the above error occurs in the c file.
It is because the c compiler can not understand extern "C".

We also thought we should support both c and cpp, so we revised it and pushed it to the [develop branch](https://github.com/ROBOTIS-GIT/OpenCR/commit/7884d0f5cd73304d09f31490e20f1f94f36cfc42#diff-75c97f7fe1dd4170e48788d08e9db521R2).
It seems to be officially released this week.

Thanks!

EdXian, 2018-01-24:
Hi @OpusK ,
BTW , Can we use the function `osDelayUntil` in our code ? It seems that the delayuntil function was not enable in `FreeRTOSConfig.h`. 
Thank you very much

OpusK, 2018-01-24:
As you can see, osDelayUntil is not available by default. We are using osDelay.
If you want, you must set INCLUDE_vTaskDelayUntil to 1 in FreeRTOSConfig.h.

closed 2018-01-24 by EdXian