ROBOTIS-GIT/OpenCR #64: How do I use I2C library ?

EdXian, 2018-01-25:


OpusK, 2018-01-26:
Hi, @EdXian 

Wire.h is not yet implemented.
Therefore, it can not be used even if the corresponding Define is used.
And the error is that there is no Wire.h in the board package. (Because it's not implemented yet)

I will discuss with the team about Wire.h support.

EdXian, 2018-01-26:
Hi @OpusK ,
Thank you for your help!
I will use the library `i2Cdev` directly . Can you give me some example about I2Cdev?
Thanks again.

OpusK, 2018-01-26:
There are no official examples using I2CDev. 
I will check to see if there is an internally implemented source.

EdXian, 2018-01-26:
HI @OpusK ,
Ok , Thanks again!

OpusK, 2018-01-26:
Hi, @EdXian 

It is not an example of OpenCR, but there is two examples using I2CDev to links below.

- [arduino_examples_oled_lcd / OpenCM9.04](https://github.com/ROBOTIS-GIT/OpenCM9.04/wiki/arduino_examples_oled_lcd)
- [arduino_examples_mpu6050_dmp / OpenCM9.04](https://github.com/ROBOTIS-GIT/OpenCM9.04/wiki/arduino_examples_mpu6050_dmp)

I think it is some useful reference to how I use the API of I2CDev.

closed 2018-02-02 by OpusK