ROBOTIS-GIT/OpenCR #65: OpenCR Upload Fail

yu1218jong, 2018-01-26:
Hi team ,
l have the same error like this issue
https://github.com/ROBOTIS-GIT/turtlebot3/issues/108

so I update openCR as following: 
Open Arduino -> Toos -> Board: -> Board Manager… -> Update (v 1.0.15)

However,  I get following error when I upload tuetlebot_core code, Do you have any advice how to fix this issue?
`Sketch uses 174064 bytes (22%) of program storage space. Maximum is 786432 bytes.
Global variables use 45372 bytes of dynamic memory.
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_319984/turtlebot3_core.ino.bin 
file size : 171 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 0.955000 sec
flash_write : 0 : 2.264000 sec 
CRC Fail : 0xF020 : 1199B30, 0 0.627000 sec
[FAIL] Download `

OpusK, 2018-01-26:
This is caused by a CRC error.
Is this problem still happening?

Hi, @yu1218jong 

I can not figure out the exact cause, but I suggest the following two ways.

1. Using the [board's recovery function](https://github.com/ROBOTIS-GIT/OpenCR/wiki/Bootloader#firmware-recovery-mode).
2. Burn the boot loader again through Arduino IDE (Burn bootloader / Programmer : DFU_UTIL).

OpusK, 2018-01-26:
Hi, @yu1218jong 

This is caused by a CRC error.
Is this problem still happening?

I can not figure out the exact cause, but I suggest the following two ways.

1. Using the [board's recovery function](https://github.com/ROBOTIS-GIT/OpenCR/wiki/Bootloader#firmware-recovery-mode).
2. Burn the boot loader again through Arduino IDE (Burn bootloader / Programmer : DFU_UTIL).

yu1218jong, 2018-01-26:
Hi OpusK, 
I burn the boot loader again through Arduino IDE, but it still happening.
However, if I use v 1.0.13, it doesn't have this problem.

OpusK, 2018-01-26:
Well, that's weird.
We have not upgraded the boot loader.

It should be the same as the CRC value of 0x1199B30.
But there's a different value.
Is this value being changed each time?
(I just uploaded it when I tested it now.)

yu1218jong, 2018-01-26:
No, this value always is 1199B30

OpusK, 2018-01-26:
Oh, I guess my explanation was not enough.
My question was the value of 0xF020.

yu1218jong, 2018-01-26:
Oh, this value is always 0xF020

yu1218jong, 2018-01-26:
Hi OpusK,

I tried to upload "turtlebot3_waffle->turtlebot3_core", and it succeed to upload.
And then I tried to upload "turtlebot3_burger->turtlebot3_core" again, but it still failed.
So I compared the differences between "turtlebot3_waffle->turtlebot3_core" and "turtlebot3_burger->turtlebot3_core".
In "turtlebot3_burger->turtlebot3_core_config.h", 
if I changed 
> #define INIT_LOG_DATA "This core is compatible with TurtleBot3 Burger"

to
> #define INIT_LOG_DATA "This core is compatible with TurtleBot3 Waffle or Waffle Pi"

It can be successfully uploaded.
Do you have any advice about this problem?

OpusK, 2018-01-26:
0xF020 is ERR_TIMEOUT.
If the response to the crc calculation is not received within 500ms from OpenCR, this error will occur.
It is not clear whether this is due to slow crc calculations.

Actually, the time when the CRC calculation succeeded is about 0.006 sec.

But, if it is due to a slow crc calculation, increasing the Timeout time (eg 500ms -> 2000ms) in [this code](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_ld/opencr_ld.c#L1008) may solve the problem.

---

While I was writing the above, I saw your new issue.
It seems to me that there is something more complicated than the problem caused by the constant change.
In this regard, it may take some time to debug.

OpusK, 2018-01-31:
Hi, @yu1218jong 

As mentioned previously, it was determined that the data for the CRC request / response was missing in the middle.

Therefore, we temporarily added the function by retrying the loader's CRC check (maximum 3 times).

This feature will be included in a future release and is now available for download in [the develop branch](https://github.com/ROBOTIS-GIT/OpenCR/tree/79f3f8986d17c2c213e9176854380b76411dd5a3).
Please use if necessary.

Thanks :)

closed 2018-01-31 by OpusK

hansenmaster, 2018-11-27:
> Hi, @yu1218jong
>
> As mentioned previously, it was determined that the data for the CRC request / response was missing in the middle.
>
> Therefore, we temporarily added the function by retrying the [...]

Hi, i have the problem 0xf020. Where can I change the timeout value? I can't open the link

hansenmaster, 2018-11-29:
> Hi, @hansenmaster
> 
> You can create a loader by modifying the source from the link below.
> 
>     * [opencr_ld](https://github.com/ROBOTIS-GIT/OpenCR/tree/develop/arduino/opencr_develop/opencr_ld)

Hi thank you for the reply, fortunately i solved the 0xf020 error with firmware reset using button on OpenCR :D