ROBOTIS-GIT/OpenCR #68: Set.setpwm adafruit library arduino

Arabmoney, 2018-02-01:
Hello there, 

I`m currently working on a project where i need to control the LED brightness with PWM through adafruit library. My issue is that i`m using this command here "set.setPWM (channelnum, time on, time off) 
so what i need is to use a time off to set a value of PWM in order to get the right brightness from the LEDs. But I have a sensor as a feedback where i use it to adjust the brightness, so when it is brighter than usual, or dimmer than the required, the PWM need to  increment or decrements. So how do i use the command set.setPWM ( 7, 0, 1865) to adjust the PWM ? 
`if( lux < 5483) `
 pwm.setPWM(7, 0, 1865++);
else   
{
 pwm.setPWM(7, 0, 1865--);

OpusK, 2018-02-01:
Hello, @Arabmoney 

I do not know what the setPWM () function used in adafruit is, but you can also use the analogWrite () function to adjust the brightness of the LED (please see [this example](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/02.%20Digital/b_PWM/b_PWM.ino))

However, this function has 8 bits of resolution. 
Therefore, if you want higher resolution, please refer to [this issue](https://github.com/ROBOTIS-GIT/OpenCR/issues/57#issuecomment-362123098).

Arabmoney, 2018-02-01:
Thanks for the quick reply. 

I`m a bit concerned about this method as i have an external pwm controller 12channels. Now i do get it each channel working, but the only issue is adjusting one channel only to regulate a temperature or an LED.
 
[
#include <Wire.h>
#include <BH1750.h>
#include <Adafruit_PWMServoDriver.h>
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();

BH1750 lightMeter;

//SDA connected to pin Analog 4
//SCL connected to pin Analog 5
//Gnd >>>Gnd  
 int pwmint = 2048;

void setup()
{
pwm.begin();
pwm.setPWMFreq(1600); 
Serial.begin(115200);
Wire.begin();
lightMeter.begin();
Serial.println(F("BH1750 Test"));
 pwm.setPWM(7, 0, pwmint);
}
void loop()
{

  uint16_t lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  delay(500);
if( lux < 5483) 
{
  pwmint--;
 pwm.setPWM(7, 0, pwmint);

}
else   
{
  pwmint++;
 pwm.setPWM(7, 0, pwmint);
//pwm.setPWM(7, 0, 4096);
//pwm.setPWM(8, 4095,0 );
}
}

}

OpusK, 2018-02-02:
Well ...
As for how to use Adafruit's library, you can get accurate and quick answers in the official forum.
We do not know about the Adafruit library.

closed 2018-02-02 by OpusK