ROBOTIS-GIT/OpenCR #69: UART and parallax DHB10 motor controller

bberrevoets, 2018-02-04:
Hello,
Thanks for the quick response on issue #66 , but i still cannot get it to work.
[image]
I have connected the board to UART2, but i am not able to communicate with the motor controller, see (https://www.parallax.com/product/28231) Can you give me a hint to connect the wires and an example schetch to get this to work? or is this setup not possible.

Kind regards,
Bert Berrevoets

OpusK, 2018-02-05:
Hello, @bberrevoets 

Have you been testing the communication with the PC with the code left on the last issue?
If it is normal communication with PC, it is not a problem of OpenCR hardware.

Is the baud rate with the controller you are using set correctly?
In the datasheet, it was explained as follows.

> Under a default configuration, connect a bidirectional serial port at 19,200 kilobaud, to CH1. For different
baud rates or separate transmit and receive pins, see the BAUD and RXPIN commands.

bberrevoets, 2018-02-07:
Hello, @OpusK 
I can control the motor board from a PC or from a Arduino board, is working ok, i also did for to be sure resetting to default setting by shortening the aux 1&2.
I do use the software in the file attached to the post.
But i does not respond.

Please can you help me? The rest i working perfectly, the ping sensors i attach are Ok, only the uart is not working.

[dhb10.zip](https://github.com/ROBOTIS-GIT/OpenCR/files/1704306/dhb10.zip)

OpusK, 2018-02-08:
umh ...

> I have connected the board to UART2,

You said you are using UART2. Is that correct?
But, the code uses Serial. This is USB.

If you want to use UART2, you must use Serial2. (Please see answer #66.)

bberrevoets, 2018-02-08:
Hello @OpusK ,
You are correct i forget to include one file, that is my dhb10_constants.h file, in that file is one line like the following.

#define DHB10_SERIAL Serial2

and that is used by the Dhb10 class. The serial communication from the main is working to my terminal, but it is the communication in the Dhb10 class that is not working.
That is the code like  DHB10_SERIAL.print etc.

OpusK, 2018-02-08:
Have you tested it directly with your PC using code from # 66?
In my opinion, it seems like you need to clarify this.

Because it is a simple UART, if it works in Arduino, it should also work without problems.
Therefore, you need to make sure that your hardware is working properly.

OpusK, 2018-03-09:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2018-03-09 by OpusK