ROBOTIS-GIT/OpenCR #70: Waffle does not startup until pressing the reset button (only when the device is connected to SBC via USB)

yoshimalucky, 2018-02-08:
In the case of connecting the brand new OpenCR to SBC via USB, waffle does not startup its turtlebot3_core program inside until pressing the reset button.

I did not see this issue with my OpenCR packaged in my turtlebot3 burger.

routiful, 2018-02-13:
Hello @yoshimalucky 

Nice to see you again :)
Which version do you use? I tested waffle firmware version 1.0.16 is working fine in my TB3 waffle.
Furthermore, If you want to reset your opencr, please check this link [OpenCR burn bootloader](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#update-bootloaderlinux)

Thanks
Darby

OpusK, 2018-03-09:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2018-03-09 by OpusK