ROBOTIS-GIT/OpenCR #73: 1.0.17  burn bootloader error

objectbuild, 2018-02-26:
arduino 1.8.1

error messages -----------------------------------------------
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Cannot open DFU device 0483:df11
No DFU capable USB device available
부트로더 굽는중 에러발생
C:\Users\gun\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-ar: unable to rename 'core\core.a'; reason: File exists

exit status 1
보드 OpenCR Board 컴파일 에러.

OpusK, 2018-02-26:
안녕하세요.

OpenCR의 보드 버전을 알 수 있을까요? (보드 매니저에 표기된)

objectbuild, 2018-02-26:
아. 확인해보니
같은버전이라 버닝이 되지 않았습니다.
감사합니다.

closed 2018-02-26 by OpusK