ROBOTIS-GIT/OpenCR #74: Not move motor when we push button.

cut-ter, 2018-02-27:
Our motor and opencr was broken.
Because we bought new motor and opencr. 

We check motor function by this(https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino) firmware and R+Manager2.
At that time we check worked motor.

we burned firmware here(https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino).

We pushed button but motor is not move.
We checked button function and we checked it by log output.

```diff
537a541,542
>   sprintf(log_msg, “1”);
>   nh.loginfo(log_msg);
544a550,551
>   sprintf(log_msg, “2");
>   nh.loginfo(log_msg);
>   
```

Button and Motor is worked but not move motor when push button.

routiful, 2018-02-27:
Hello :)

You have to check Dynamixel.
Please following below link

[turtlebot3_setup_motor](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#turtlebot3-setup-dynamixels)

Thanks
Darby

cut-ter, 2018-02-27:
Thanks we worked.

We have problem when we execute it.
```
Succeeded to open the port!
 - Device Name : /dev/OpenCR
 - Baudrate    : 57600
Start turtlebot3 setup motor
 
 
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> Do you really want to setup ? y/n : yes
setup.... left
Find Motor...
    setbaud : 57600
This function is deprecated. Use 'Serial.print()' and 'getRxPacketError()' instead
[TxRxResult] There is no status packet!
    setbaud : 1000000
    ... SUCCESS
    [ID:1]
    found motor
Setup Motor Left...
This function is deprecated. Use 'Serial.print()' and 'getRxPacketError()' instead
[TxRxResult] There is no status packet!
Fail to write!
This function is deprecated. Use 'Serial.print()' and 'getRxPacketError()' instead
[TxRxResult] There is no status packet!
Fail to write!
This function is deprecated. Use 'Serial.print()' and 'getRxPacketError()' instead
[TxRxResult] There is no status packet!
Fail to write!
    ok
 
 
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> 
```

But we solved by  R+Manager refer below code.
```
   write(portHandler, packetHandler2, tb3_id,64, 1, 0);
   write(portHandler, packetHandler2, tb3_id, 7, 1, 1);
   tb3_id = 1;
   write(portHandler, packetHandler2, tb3_id, 8, 1, 3);
   portHandler->setBaudRate(1000000);
   write(portHandler, packetHandler2, tb3_id, 10, 1, 0);
   write(portHandler, packetHandler2, tb3_id, 11, 1, 1);
   CMD_SERIAL.println(”    ok”);

   write(portHandler, packetHandler2, tb3_id,64, 1, 0);
   write(portHandler, packetHandler2, tb3_id, 7, 1, 2);
   tb3_id = 2;
   write(portHandler, packetHandler2, tb3_id, 8, 1, 3);
   portHandler->setBaudRate(1000000);
   write(portHandler, packetHandler2, tb3_id, 10, 1, 1);
   write(portHandler, packetHandler2, tb3_id, 11, 1, 1);
   CMD_SERIAL.println(”    ok”);
```

closed 2018-02-27 by cut-ter

routiful, 2018-02-27:
Thanks. we will update it next release :)