ROBOTIS-GIT/OpenCR #75: 5VDC power only outputting 900mA

yycho0108, 2018-03-03:
### Main Issue

I don't know if this is the proper place to raise issues about the OpenCR Hardware.
I recently ordered Turtlebot Burger from Dabit Industries, and fitted it ODROID XU4 as opposed to the standard Raspberry Pi 3.

Unfortunately, it got stuck in a loop of restarting, so I figured it must be an issue with the power supply.
When I checked the current on the 5VDC power output, it was only outputting 900mA ( according to the specifications, it should be able to supply 4A@5V; my battery was fully charged).

I had to order a new OpenCR board, as I didn't believe ROBOTIS would accept repairs/replacements for defects on a component from a part ordered from another distributor. While the shipment arrives, it would be great if you could double check on the fact that the board **could** supply 4A@5V at all. It was certainly unable to do so from the get-go.

### Reviving the board
On a separate note, in the process of debugging the issue, I tried to probe the "power out" pins beside the barrel-jack connector, and saw a spark. Presumably, that broke the board's power distribution, as the board will no longer accept input power from the SMP, nor the LiPo battery (it can only be turned on by power input over USB, which is insufficient to power the actuators). Is it possible the revive the board, in some way?

yycho0108, 2018-03-05:
Nothing else is plugged into the OpenCR except for the Dynamixels, as you mentioned, and the ODROID XU4 (5VDC for power, and USB connection for communication)

Can confirm that the ODROID XU4 runs fine on a separate power supply that provides 5V@4A.

ROBOTIS-Will, 2018-03-06:
5V converter IC is XL4005 which can provide 5A at maximum so it should be able to supply enough power.  
If you need to get it repaired, you can get RMA for any defects on our product purchased from our distributors.  
Please visit our [repair center](http://en.robotis.com/model/page.php?co_id=repaircenter) :)

yycho0108, 2018-03-15:
Update: it turned out to be a cable issue on my side. Thank you for all the help!

closed 2018-03-15 by yycho0108