ROBOTIS-GIT/OpenCR #79: motors getting locked when using with opencr (XM430-w210-t)

allanis79, 2018-03-13:
Hello all,

I recently bought turtleobt3, it came with dynamixel motors Xm430-w210-t, intel joule and openCR board. I followed the tutorial and done all the steps. 

when i launched turtlebot3 bringup and lauched teleoperation it didn't show any errors in the terminal, but the turtlebot didn't move at all when i pressed the keys. I can see the output on /odom topic. 

It looks like motors got locked. I tired rotating them manually but they didn't rotate. I didn't force, i thought i might break something inside.

Can anyone help me what's happening?

Thanks

routiful, 2018-03-13:
Hi :)

Firmware version 1.1.0 has bug to control dynamixel.
We hotfixed it and now you can download v.1.1.1.
We apologize for any inconvenience.

Thanks
Darby

allanis79, 2018-03-14:
I am not able to upload the firmware. I am getting the following error:

allanis79, 2018-03-14:
I uploaded the new firmware v.1.1.1. My motors were moving before i uploaded the firmware, now they got locked. I think there is something wrong with the firmware.

routiful, 2018-03-15:
Hello :)

I checked it firmware v.1.1.1. It has not problem to control dynamixel.
Would you check your dynamixel setup?

[FAQ](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#turtlebot3-setup-dynamixels)

Thanks

allanis79, 2018-03-15:
Thanks, my motors are moving now. But still i can't get the robot working. when I run the following command, I am getting these errors:

`roslaunch turtlebot3_bringup turtlebot3_core.launch`

`SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

auto-starting new master
process[master]: started with pid [2129]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 98ea3e6c-27fc-11e8-9faf-a0c5894b6b8a
process[rosout-1]: started with pid [2142]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [2145]
[INFO] [1521082650.790351]: ROS Serial Python Node
[INFO] [1521082650.802078]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1521082653.106244]: Note: publish buffer size is 1024 bytes
[INFO] [1521082653.106850]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1521082653.111552]: Setup publisher on version_info [turtlebot3_msgs/VersionInfo]
[INFO] [1521082653.116422]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1521082653.123235]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1521082653.148304]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1521082653.153266]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1521082653.158437]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1521082653.163153]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1521082655.031962]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1521082655.039057]: Note: subscribe buffer size is 1024 bytes
[INFO] [1521082655.039888]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1521082655.047288]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1521082655.056359]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1521082655.064899]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1521082656.095401]: --------------------------
[INFO] [1521082656.097916]: Connected to OpenCR board!
[INFO] [1521082656.099867]: This core(v1.1.1) is compatible with TurtleBot3 Waffle or Waffle Pi
[INFO] [1521082656.101626]: --------------------------
[ERROR] [1521082682.914587]: Lost sync with device, restarting...
[ERROR] [1521082712.917347]: Lost sync with device, restarting...
`

Can you help me?

routiful, 2018-03-15:
Hello :)

We find some bug in Firmware.
Would you upload the latest version (1.1.2) to your OpenCR?

Best regard
Darby

allanis79, 2018-03-16:
Thanks, It's working. I think you should should do better job at writing the documentation. It took me more than 1 week to get turtlebot3 running. Anyway, thanks.

routiful, 2018-03-18:
Sorry for inconvenience.
We are going to write the documentation more better everyday.
Thanks you for your kindly advise :)

If you want to any question about TB3, please come again

Thanks
Darby

closed 2018-03-18 by routiful