ROBOTIS-GIT/OpenCR #85: Arduino: Compile sketches without warnings

KurtE, 2018-04-26:
Sorry not sure if this is right place or if you are wanting issues raised like this.

I typically try to build my applications to be able to do builds without any compiler warnings as often times warnings turn out to be bugs I have not ran into yet... 

So if you choose OpenCR as the board type, and turn on compile warnings: 
File->Preferences (or CTRL-,) 
And turn on options for Vebose output Compile, and compiler Warnings ALL,
And then try to build a program like the Example a_Serial_HelloWorld I am currently seeing lots of Warnings:

Note: I am running Arduino 1.8.5 and OpenCR 1.1.2... Windows 10 64bit
Output shown during Build:  

```

D:\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino-1.8.5\hardware -hardware C:\Users\kurte\AppData\Local\Arduino15\packages -hardware C:\Users\kurte\Documents\Arduino\hardware -tools D:\arduino-1.8.5\tools-builder -tools D:\arduino-1.8.5\hardware\tools\avr -tools C:\Users\kurte\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino-1.8.5\libraries -libraries C:\Users\kurte\Documents\Arduino\libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10805 -build-path C:\Users\kurte\AppData\Local\Temp\arduino_build_961740 -warnings=all -build-cache C:\Users\kurte\AppData\Local\Temp\arduino_cache_852849 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_tools.path=C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_tools\1.0.0 -prefs=runtime.tools.opencr_gcc.path=C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2 -verbose C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\libraries\OpenCR\examples\03. Communication\a_Serial_HelloWorld\a_Serial_HelloWorld.ino
D:\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware D:\arduino-1.8.5\hardware -hardware C:\Users\kurte\AppData\Local\Arduino15\packages -hardware C:\Users\kurte\Documents\Arduino\hardware -tools D:\arduino-1.8.5\tools-builder -tools D:\arduino-1.8.5\hardware\tools\avr -tools C:\Users\kurte\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino-1.8.5\libraries -libraries C:\Users\kurte\Documents\Arduino\libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10805 -build-path C:\Users\kurte\AppData\Local\Temp\arduino_build_961740 -warnings=all -build-cache C:\Users\kurte\AppData\Local\Temp\arduino_cache_852849 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_tools.path=C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_tools\1.0.0 -prefs=runtime.tools.opencr_gcc.path=C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2 -verbose C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\libraries\OpenCR\examples\03. Communication\a_Serial_HelloWorld\a_Serial_HelloWorld.ino
Using board 'OpenCR' from platform in folder: C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2
Using core 'arduino' from platform in folder: C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2
Detecting libraries used...
"C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10805 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/bsp/opencr" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/bsp/opencr/include" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw/driver" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw/usb_cdc" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\cores\arduino" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR" "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740\sketch\a_Serial_HelloWorld.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10805 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/bsp/opencr" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/bsp/opencr/include" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw/driver" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw/usb_cdc" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\cores\arduino" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR" "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740\sketch\a_Serial_HelloWorld.ino.cpp" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740\preproc\ctags_target_for_gcc_minus_e.cpp"
"D:\arduino-1.8.5\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10805 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/bsp/opencr" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/bsp/opencr/include" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw/driver" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/hw/usb_cdc" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\cores\arduino" "-IC:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR" "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740\sketch\a_Serial_HelloWorld.ino.cpp" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740\sketch\a_Serial_HelloWorld.ino.cpp.o"
In file included from C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\cores\arduino/USBSerial.h:37:0,

                 from C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.1.2\variants\OpenCR/variant.h:34,

[... cut ...] d2072f71d795ea5f7a61b11c4e58.a" -Wl,--no-whole-archive -Wl,--end-group
"C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-objcopy" -O binary  "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740/a_Serial_HelloWorld.ino.elf" "C:\Users\kurte\AppData\Local\Temp\arduino_build_961740/a_Serial_HelloWorld.ino.bin"
Sketch uses 37152 bytes (4%) of program storage space. Maximum is 786432 bytes.
Global variables use 55372 bytes of dynamic memory.
```

OpusK, 2018-04-27:
Hi, @KurtE 

I totally agree with your opinion.
I think it is right to remove warnings as much as possible.

So, I will try to find these warnings and remove them.

Thank you for your contributions.

OpusK, 2018-04-30:
@KurtE,
I have modified the code for All warning except for a few things below.

- MadgwickAHRS
- Adafruit's libraries
- Dynamixel Workbench
- FreeRTOS (cmsis_os, etc..)

In the case of Dynamixel Workbench, this issue will be communicated to the team that manages the code and will be modified or not, depending on the team's decision.
Other libraries are not modified because they use external libraries. (Exceptionally, the HAL library has been modified since it applies to all compilations)

If you have other improvements, please leave issues.
Thanks :)

closed 2018-04-30 by ROBOTIS-Will

KurtE, 2018-04-30:
Thanks,

 

Hopefully Dynamixel Workbench will be updated as well, although I don’t typically use it except to maybe try out a thing or two.

 

I know with Teensy processors (PJRC), we were able to get Adafruit to take some Pull Requests to update some of their libraries in the past.

Who knows maybe at some point I will ty out some of them and see if they will take the updates…  But maybe at some point I need to get

A second OpenCR board, to experiment on outside of the Turtlebot 3… 

 

Again Thanks.

 

P.S – In some of these cases do you prefer Issues or for example the issue about platform.txt I could have setup to issue a Pull Request. 

But not sure if you would prefer it based on Master or develop branch, I am guessing the later.

commit 92ded038 (2018-04-27): Fixed #85 (DynamixelSDK examples)
commit 89068530 (2018-04-27): Fixed #85 (Some examples in Etc)