ROBOTIS-GIT/OpenCR #86: Arduino: Does not support Sketch->Export compiled binary command

KurtE, 2018-04-29:
With most Arduino board types, you can use this command to do a build, and then have put a copy of the output binary into your sketch folder.

This can be useful for example if you wish to build the Turtlebot3 Arduino code, and then use something like RCP or WinSCP or... to copy the binary to your SBC board, where you can hopefully use the commands like opencr_ld_shell_arm (or... _x86) to then have the SBC machine 

I am pretty sure you just need to add a couple of lines to the platform.txt file.

```
## Create hex or binary
recipe.objcopy.hex.pattern="{compiler.path}{compiler.elf2hex.cmd}" {compiler.elf2hex.flags} {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.bin"

## Save hex
recipe.output.tmp_file={build.project_name}.bin
recipe.output.save_file={build.project_name}.{build.variant}.bin
```

Note: I only added the Save hex lines, the create hex or binary are already there.  I actually copied those lines from the Arduino update for the SAM boards. 

When I tried it out. With a copy of the Turtlebot3_core (for waffle), that I renamed to: turtlebot3_core_unchanged

It create the file: turtlebot3_core_unchanged.ino.OpenCR.bin
in the sketch directory.   I have not tried it out yet on the Turtle but hopefully it works.

KurtE, 2018-04-29:
Update: As I mentioned, it created the file:   I then used winscp  copy it down to my in this case UP board (x86), I then tried using the ./update script that is used to download new firmware...  And it does not like that file:

Note: I ran the default waffle one to begin and it ran fine:

```
kurt@kurt-UP-Turtle:~/opencr_update$ ./update.sh /dev/ttyACM0 waffle.opencr
x86_64
amd64
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main
[  ] file name          : waffle.opencr
[  ] file size          : 172 KB
[  ] fw_name            : waffle
[  ] fw_ver             : 1.1.2
[OK] Open port          : /dev/ttyACM0
[  ]
[  ] Board Name         : OpenCR R1.0
[  ] Board Ver          : 0x17020800
[  ] Board Rev          : 0x00000000
[OK] flash_erase        : 0.99s
[OK] flash_write        : 0.93s
[OK] CRC Check          : 118B28A 118B28A , 0.005000 sec
[OK] Download
[OK] jump_to_fw
kurt@kurt-UP-Turtle:~/opencr_update$ ls
burger.opencr                   opencr_ld_shell_x86    released_1.1.2.txt                        waffle.opencr
burger_turtlebot3_core.ino.bin  opencr_update          turtlebot3_core_unchanged.ino.OpenCR.bin  waffle_turtlebot3_core.ino.bin
opencr_ld_shell_arm             opencr_update.tar.bz2  update.sh
kurt@kurt-UP-Turtle:~/opencr_update$ ./update.sh /dev/ttyACM0 waffle_turtlebot3_core.ino.bin
x86_64
amd64
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main
[  ] file name          : waffle_turtlebot3_core.ino.bin
[  ] file size          : 170 KB
[NG] not opencr fw
[  ] file name          : waffle_turtlebot3_core.ino.bin
[  ] file size          : 170 KB
[NG] not opencr fw
[  ] file name          : waffle_turtlebot3_core.ino.bin
[  ] file size          : 170 KB
[NG] not opencr fw
kurt@kurt-UP-Turtle:~/opencr_update$
```

So maybe need to figure out if there is some step missing or should I be trying a different path?  Like maybe install the Arduino IDE on the up board and either try it's upload procedure hopefully from command line? Still investigating

OpusK, 2018-04-30:
Hi, @KurtE.

For OpenCR, the BOOTLOADER demand TAG information for an Application.
To do this, opencr_ld creates an additional binary with TAG information. (This is .opencr file)
The binary file you get from Arduino is a binary with no tag information, so opencr_ld returns an error.

If you want to use the update.sh file, first create an .opencr binary file via opencr_ld and run update.sh with the generated file.

Here is how to create an .opencr binary using opencr_ld_shell.
>>  $ opencr_ld_shell_x86 make "binary file" "firmware_name" "firmware_version"
(eg. $ opencr_ld_shell_x86 make  waffle_turlebot3_core.ino.bin waffle_custum.opencr 1.0.0)

ROBOTIS-Will, 2018-04-30:
@KurtE 
What I do for releasing the binary file, 
1. Use "verify" on Arduino IDE to build the sketch 
2. Find the "project".ino.bin file from "C:\Users\[username]\AppData\Local\Temp\arduino_build_xxxxxx" folder.  
3. Use "opencr_ld_shell_x86" to add TAG information to the binary file.
Please try the binary file acquired from above method and let us know how does it work.
Thank you.

KurtE, 2018-04-30:
Thanks @OpusK and @ROBOTIS-Will 

That appears to work.   - Might be good to add some comments about this in the e-manual about how to update the OpenCR. 

I still think it would help to add the sketch->Export... command as it makes it easier to not have to search through your temp files to find where it put the files. 

Thanks again!

ROBOTIS-Will, 2018-04-30:
@KurtE 
Thank you for the idea!
I'll transfer some of materials regarding firmware update in TurtleBot3 to OpenCR.

closed 2018-04-30 by ROBOTIS-Will