ROBOTIS-GIT/OpenCR #88: digitalWriteFast would be nice to have

KurtE, 2018-04-30:
As I discussed on the Forum, you can get a lot better performance on digital IO if you directly talk to the registers.  This is if you are talking directly to specific constant IO pin... 

http://en.robotis.com/service/forum_view.php?bbs_no=2406779&page=1&save_sca=&sca=&save_stx=&stx=&sfl=

But it is pretty messy and problematic if the user has to do it manually.  On AVR arduino as well as Teensy, they added another set of functions digitalWriteFast (and digitalReadFast) which for fixed constant pins goes directly to the registers... 

Example code: function showing different code types of code:

```
GPIO_TypeDef *pin1_port;
uint32_t  pin1_set_mask;
uint32_t  pin1_clear_mask;
uint16_t loop_count;

void setup() {
  // put your setup code here, to run once:
  while (!Serial && millis() < 4000) ; 
  Serial.begin(115200);
  Serial.println("Start Test");
  pinMode (0, OUTPUT);
  for(int i=0; i < 5; i++) {
    digitalWrite(0, HIGH);
    delay(1);
    digitalWrite(0, LOW);
    delay(1);
  }
  pinMode (1, OUTPUT);
  for(int i=0; i < 5; i++) {
    digitalWrite(1, HIGH);
    delay(1);
    digitalWrite(1, LOW);
    delay(1);
  }
  pinMode(2, OUTPUT);
  // Lets get and cache for pin 1... 
  pin1_port = g_Pin2PortMapArray[1].GPIOx_Port;
  pin1_set_mask = g_Pin2PortMapArray[1].Pin_abstraction;
  pin1_clear_mask = (uint32_t)g_Pin2PortMapArray[1].Pin_abstraction << 16;

}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(0, HIGH);
  digitalWrite(0, LOW);
  digitalWrite(0, HIGH);
  digitalWrite(0, LOW);
  digitalWrite(0, HIGH);
  digitalWrite(0, LOW);
  
  // Lets try the HAL way?
  pin1_port->BSRR = pin1_set_mask;
  pin1_port->BSRR = pin1_clear_mask;
  pin1_port->BSRR = pin1_set_mask;
  pin1_port->BSRR = pin1_clear_mask;
  pin1_port->BSRR = pin1_set_mask;
  pin1_port->BSRR = pin1_clear_mask;

  // put your main code here, to run repeatedly:
  digitalWriteFast(2, HIGH);
  digitalWriteFast(2, LOW);
  digitalWriteFast(2, HIGH);
  digitalWriteFast(2, LOW);
  digitalWriteFast(2, HIGH);
  digitalWriteFast(2, LOW);
}
```
That does pin 0 the normal way, pin 1 the save register names earlier and use masks and then pin 2 using digitalWriteFast. 

I have included a zip file with a first pass done for digitalWriteFast as well as this test program.  Not sure which route if any you wish to do.  AVR Arduino it was done as external library.  Teensy, he built it directly into the main core header files...  

I ran this on my opencr with Logic analyzer attached and you can see the three IO pins, where the digitalWriteFast works more or less the same as the hard coded IO pins.
[image]

[opencr_gpio_speed_test-180430a.zip](https://github.com/ROBOTIS-GIT/OpenCR/files/1961508/opencr_gpio_speed_test-180430a.zip)

I also looked at the generated code: using the command: 
PS C:\Users\kurte\AppData\Local\Temp\arduino_build_318322> C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\
opencr_gcc\5.4.0-2016q2\arm-none-eabi\bin\objdump -D .\opencr_gpio_speed_test.ino.elf > foo

When I looked at foo: 

```
8003b50 <loop>:
 8003b50:	b510      	push	{r4, lr}
 8003b52:	2101      	movs	r1, #1
 8003b54:	2000      	movs	r0, #0
 8003b56:	f005 fdc5 	bl	80096e4 <digitalWrite>
 8003b5a:	2100      	movs	r1, #0
 8003b5c:	4608      	mov	r0, r1
 8003b5e:	f005 fdc1 	bl	80096e4 <digitalWrite>
 8003b62:	2101      	movs	r1, #1
 8003b64:	2000      	movs	r0, #0
 8003b66:	f005 fdbd 	bl	80096e4 <digitalWrite>
 8003b6a:	2100      	movs	r1, #0
 8003b6c:	4608      	mov	r0, r1
 8003b6e:	f005 fdb9 	bl	80096e4 <digitalWrite>
 8003b72:	2101      	movs	r1, #1
 8003b74:	2000      	movs	r0, #0
 8003b76:	f005 fdb5 	bl	80096e4 <digitalWrite>
 8003b7a:	2100      	movs	r1, #0
 8003b7c:	4608      	mov	r0, r1
 8003b7e:	f005 fdb1 	bl	80096e4 <digitalWrite>
 8003b82:	490d      	ldr	r1, [pc, #52]	; (8003bb8 <loop+0x68>)
 8003b84:	4b0d      	ldr	r3, [pc, #52]	; (8003bbc <loop+0x6c>)
 8003b86:	4a0e      	ldr	r2, [pc, #56]	; (8003bc0 <loop+0x70>)
 8003b88:	681b      	ldr	r3, [r3, #0]
 8003b8a:	680c      	ldr	r4, [r1, #0]
 8003b8c:	6812      	ldr	r2, [r2, #0]
 8003b8e:	611c      	str	r4, [r3, #16]
 8003b90:	490c      	ldr	r1, [pc, #48]	; (8003bc4 <loop+0x74>)
 8003b92:	611a      	str	r2, [r3, #16]
 8003b94:	480c      	ldr	r0, [pc, #48]	; (8003bc8 <loop+0x78>)
 8003b96:	611c      	str	r4, [r3, #16]
 8003b98:	611a      	str	r2, [r3, #16]
 8003b9a:	611c      	str	r4, [r3, #16]
 8003b9c:	611a      	str	r2, [r3, #16]
 8003b9e:	f005 fdc9 	bl	8009734 <_ZN5Print5printEPKc>
 8003ba2:	4b0a      	ldr	r3, [pc, #40]	; (8003bcc <loop+0x7c>)
 8003ba4:	6bda      	ldr	r2, [r3, #60]	; 0x3c
 8003ba6:	6b9b      	ldr	r3, [r3, #56]	; 0x38
 8003ba8:	0411      	lsls	r1, r2, #16
 8003baa:	611a      	str	r2, [r3, #16]
 8003bac:	6119      	str	r1, [r3, #16]
 8003bae:	611a      	str	r2, [r3, #16]
 8003bb0:	6119      	str	r1, [r3, #16]
 8003bb2:	611a      	str	r2, [r3, #16]
 8003bb4:	6119      	str	r1, [r3, #16]
 8003bb6:	bd10      	pop	{r4, pc}
```
It also looked pretty similar.  Note: I had a simple Serial.print("."); between the using the registers directly and using my digitalWriteFast.  

Note: I have not looked at doing digitalReadFast yet.  But probably would be very similar.

OpusK, 2018-05-02:
Hi, @KurtE,

I checked your code and it is run well.
We have basically worked on the basic API compatibility of Arduino. However, from a performance standpoint, we thought it would be better to add more optimized APIs to the ARM core.

Therefore, I am pleased about your suggestion.
So, do you want us to add about the features of this issue? Or, do you want PullRequest as mentioned in #85 ?

If there are any improvements in the future, please leave an issue or PullRequest at any time.

Again, thanks for your contributions.

KurtE, 2018-05-03:
Right now, it would probably be better if you did the merge as it is unclear where/how you would like it.

As I mentioned on AVR, you have to include a library for this. 

On Teensy it is integrated into Core.  If I remember correctly it was in same header file that also declares digitalWrite and digitalRead.  Also probably needs a standardized set of comments.  I can do this if desired, but I usually prefer to do PR on simple fixes or extensions.

OpusK, 2018-05-04:
As I answered #85, you can PR to develop branch. (from develop branch based)
As you mentioned about the Teensy board, we also like to be integrated into Core.

I may merge myself into the .zip file you uploaded, but I think it would be better to leave your contribution.

Comments can be added or modified after PR, so please feel free to PR as you like.

Thanks :)

KurtE, 2018-05-04:
I put up first version of a Pull request in: 
https://github.com/ROBOTIS-GIT/OpenCR/pull/90

Note: the user still has to include: #include <digitalWriteFast.h>
On the Teensy boards it is automatic, so I am torn between simply adding the stuff to 
digital_wiring.h or add include of this file to digital_wiring.h 

Let me know if you would like me to change anything

OpusK, 2018-05-08:
merged #90

closed 2018-05-08 by OpusK