ROBOTIS-GIT/OpenCR #89: Opencr Board without USB - Usable ?

lucasilvassis, 2018-05-04:
Hi guys, just bought a turtlebot 3 and noticed the USB port is detached from the board, some PCB paths were beyond recover by soldering. Can I still use the board with a Serial/USB adapter or other device? If so, witch connections should I make ?

lucasilvassis, 2018-05-04:
Hi, @OpusK this is the closer i can get with focus on my phone, sorry if its no enough.

Two pahts were fully detached and one lost amost half of it for what i can tell. 

[image]

Well, if there is no work around I will try to send it for repair, i guess it was damaged in factory/shipping and the USB port was just hanging.

OpusK, 2018-05-04:
It seems to me that repair is difficult.

OK, please contact the [repair center](http://en.robotis.com/model/page.php?co_id=repaircenter) and check if repair is possible.

closed 2018-05-11 by robotpilot