ROBOTIS-GIT/OpenCR #95: 1.1.13 incorrectly reports its version number

ihadzic, 2018-05-22:
Just upgraded to 1.1.13. Upgrade successful, but the /version_info topic still claims it's 1.1.12:

Update report:

```
./update.sh /dev/ttyACM0 waffle.opencr 
x86_64
amd64
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main 
[  ] file name   	: waffle.opencr 
[  ] file size   	: 172 KB
[  ] fw_name     	: waffle 
[  ] fw_ver      	: 1.1.3 
[OK] Open port   	: /dev/ttyACM0
[  ]
[  ] Board Name  	: OpenCR R1.0
[  ] Board Ver   	: 0x17020800
[  ] Board Rev   	: 0x00000000
[OK] flash_erase 	: 1.01s
[OK] flash_write 	: 0.89s 
[OK] CRC Check   	: 118EAE2 118EAE2 , 0.005000 sec
[OK] Download 
[OK] jump_to_fw 
```

`/version_info' topic reports this:

```
hardware: "1.0.0"
firmware: "1.1.2"
software: "1.0.0"
---

```

ROBOTIS-Will, 2018-05-23:
@ihadzic 
Thank you for reporting the bug.
FYI, OpenCR firmware and TurtleBot3 firmware have different version although they are related to each other.
The version you can see from Arduino IDE > board manager indicates the OpenCR Arduino IDE version(The latest version is 1.1.3).
TurtleBot3 firmware is included in the OpenCR Arduino IDE example and it keeps [different version labeling](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core_config.h#L49).
Although the latest TurtleBot3 firmware version is 1.1.2, the recent OpenCR update used wrong version number.
The next update will make both OpenCR and TurtleBot3 version to 1.2.0, but still both version can differ later.
Thanks.

ihadzic, 2018-05-23:
How can my code then read the OpenCR version number? I have workarounds for firmware bugs in my code (see my other bug reports) and I rely on version number read from /version_info topic to decide which workarounds to apply. If firmware version number can diverge from the OpenCR version number, then my code does not have enough information to know which firmware it is dealing with.

OpusK, 2018-05-25:
Updated.

closed 2018-05-25 by OpusK