ROBOTIS-GIT/OpenCR #96: linear acceleration on /imu topic has wrong sign

ihadzic, 2018-05-22:
When the Turtlebot is flat on the ground, not moving, the linear acceleration shows negative numbers with G-vector pointing down:

```
linear_acceleration: 
  x: -0.310362011194
  y: -0.0197340007871
  z: -9.91364383698
```

Apparently this is done by design because ACCEL_FACTOR constant in turtlebot3_sensor.h is negative. This wrong and in conflict with ROS-REP145: http://www.ros.org/reps/rep-0145.html

In section titled "Data Reporting" it reads: "When the device is at rest, the vector will represent the specific force solely due to gravity. I.e. if the body z axis points upwards, its z axis should indicate +g. This data must be in m/s^2."

Other ROS modules that use accelerometer and comply with this specification will get utterly confused. The ACCEL_FACTOR should not be negative.

ROBOTIS-Will, 2018-05-23:
@ihadzic,
Thank you for reporting the issue.
This will be applied in the next release within this week.
Thanks!

routiful, 2018-05-23:
Hi :) @ihadzic 

Thank you for your inquiry. 
You can watch the code is replaced.

https://github.com/ROBOTIS-GIT/OpenCR/blob/develop/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/turtlebot3/turtlebot3_sensor.h#L28

It will be updated soon. Thank you for your help!!

OpusK, 2018-05-25:
Updated.

closed 2018-05-25 by OpusK

routiful, 2018-05-25:
Hi again @ihadzic

We update new firmware (version 1.2.0) :) Thank you to your contribution!