ROBOTIS-GIT/OpenCR #100: Arduino upload command fail if Serial monitor is open.

KurtE, 2018-05-27:
This is probably a known issue, but thought I would mention. 

If I have the Arduino Serial monitor window open and try to do an upload command, the command will fail as it could not open the Serial port. 

With some other boards, when you issue the Upload command, they will gray out the contents of the Serial monitor window,  then when the upload completes, it automatically re-enables the Serial monitor window.  Which is nice, especially if you have Serial outputs that happen quickly.

OpusK, 2018-05-28:
As far as I know, some Arduino boards use an external chip for USB communication, and I know that this chip and MCU reset work separately.

Can you tell me what other boards you mentioned?

KurtE, 2018-05-28:
As you mentioned, some Arduino boards like the UNO, and MEGA, use an external chip to support the USB.  SOmetimes these are FTDI chips and other times they use another AVR chip like an Atmega32u2 to handle the communications. 

But some other boards have native USB support.  This includes some AVR chips like the Atmega 32u2 or 32u4 chips (Leonardo, Teensy 2 ...).

And the main ones I use, the teensy 3.x processors (PJRC), have native ARM USB support.  I know that it can be tricky, getting the timing correct.    

I don't know enough of the Arduino internals, like when you start a build, the Serial monitor goes into the inactive state (text disabled...) and when/if/how the Serial port is logically closed, such that the tools needed to do the update can open it, and then how Arduino tells the Serial monitor it now has the Comm port again...  But it is nice when it is working.

OpusK, 2018-05-28:
OK, we will proceed with this.

chcbaram, 2018-05-28:
[opencr_ld.zip](https://github.com/ROBOTIS-GIT/OpenCR/files/2043264/opencr_ld.zip)

@KurtE 

Would you try modified opencr_ld.exe for uploading without closing serial monitor?
You have to copy attached file to the folder blow.

\Arduino15\packages\OpenCR\tools\opencr_tools\1.0.0\win

I am not sure this is what you want. It's possible to upload with serial monitor begin open.
Please let me know it's working.

Thank.

KurtE, 2018-05-28:
Thanks, I tried it, and Java gave an error when I tried reloading the program when the Serial monitor was open

The good news looks like it was able to do the download, but maybe held the stream open or something so than Java error

```
C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_tools\1.0.0/win/opencr_ld.exe COM31 115200 C:\Users\kurte\AppData\Local\Temp\arduino_build_981385/OLLO_IR_Read.ino.bin 1 
opencr_ld ver 1.0.3
opencr_ld_main 

>>

file name : C:\Users\kurte\AppData\Local\Temp\arduino_build_981385/OLLO_IR_Read.ino.bin 

file size : 41 KB

Open port OK
Clear Buffer Start
Clear Buffer End
Board Name : OpenCR R1.0

Board Ver  : 0x17020800

Board Rev  : 0x00000000

>>

flash_erase : 0 : 0.980000 sec

flash_write : 0 : 0.286000 sec 

CRC OK 3F9DFA 3F9DFA 0.001000 sec

[OK] Download 

jump_to_fw 

processing.app.SerialException: Error opening serial port 'COM31'.
	at processing.app.Serial.<init>(Serial.java:149)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$4.<init>(SerialMonitor.java:101)
	at processing.app.SerialMonitor.open(SerialMonitor.java:101)
	at processing.app.AbstractMonitor.resume(AbstractMonitor.java:104)
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2241)
	at processing.app.Editor.access$2200(Editor.java:79)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2207)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - COM31; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.<init>(Serial.java:138)
	... 8 more
Error opening serial port 'COM31'.

```

But I could then close the window and reopen.

chcbaram, 2018-05-28:
[opencr_ld.zip](https://github.com/ROBOTIS-GIT/OpenCR/files/2046326/opencr_ld.zip)

Thank you for replay. 

It's working on my computer but I think there is some time difference on many computers.
So I modified that it waits to find the comport before exiting download program up to 3000ms

Please try attached file. Let me know it's working.

Thanks.

KurtE, 2018-05-29:
Sounds good. 

This one worked on my machine. :D 

Thanks

closed 2018-05-29 by OpusK