ROBOTIS-GIT/OpenCR #101: IMU --time error

gavinwangpj, 2018-05-31:
Hello everyone,
    I'm use turtlebot3 waffle pi, and ros-kinect, ubuntu 16.04;opencr 1.1.2
    but the time of imu has some error;
 -----------------------------------------------------------------------
time                                          .seq       stamp.secs     stamp.nsecs
2018/05/30/10:16:27.790040 | 76052 | 1527646585 | 780853009
2018/05/30/10:16:27.815596 | 76053 | 1527646585 | 792855807
2018/05/30/10:16:27.821879 | 76054 | 1527646585 | 807853008
2018/05/30/10:16:27.831077 | 76055 | 1527646585 | 819853009
2018/05/30/10:16:27.862529 | 76056 | **1527646589 | 844183806**
2018/05/30/10:16:27.866657 | 76057 | 1527646589 | 859181008
2018/05/30/10:16:27.870838 | 76058 | 1527646589 | 871181008
2018/05/30/10:16:27.886552 | 76059 | 1527646589 | 883183815
2018/05/30/10:16:27.890362 | 76060 | 1527646589 | 898181009
2018/05/30/10:16:27.894518 | 76061 | 1527646589 | 910181008
2018/05/30/10:16:27.916989 | 76062 | 1527646589 | 922183820
2018/05/30/10:16:27.921492 | 76063 | 1527646589 | 937181008
2018/05/30/10:16:27.942004 | 76064 | 1527646589 | 949181008
2018/05/30/10:16:27.957147 | 76065 | 1527646589 | 961183820
2018/05/30/10:16:27.960795 | 76066 | 1527646589 | 976181008
2018/05/30/10:16:27.971516 | 76067 | 1527646589 | 988181008
2018/05/30/10:16:27.998406 | 76068 | 1527646590 | 183819
2018/05/30/10:16:28.002351 | 76069 | 1527646590 | 15181009
2018/05/30/10:16:28.011485 | 76070 | 1527646590 | 27181008
2018/05/30/10:16:28.041884 | 76071 | 1527646590 | 39183820
2018/05/30/10:16:28.045946 | 76072 | 1527646590 | 54181008
2018/05/30/10:16:28.050558 | 76073 | 1527646590 | 66181009
2018/05/30/10:16:28.073170 | 76074 | 1527646590 | 78183817
2018/05/30/10:16:28.077685 | 76075 | 1527646590 | 93181008
2018/05/30/10:16:28.087635 | 76076 | 1527646590 | 105181009
2018/05/30/10:16:28.110887 | 76077 | 1527646590 | 117183820
2018/05/30/10:16:28.115882 | 76078 | 1527646590 | 132181008
2018/05/30/10:16:28.128186 | 76079 | 1527646590 | 144181009
2018/05/30/10:16:28.153341 | 76080 | 1527646590 | 156183820
2018/05/30/10:16:28.157780 | 76081 | 1527646590 | 171181009
2018/05/30/10:16:28.165341 | 76082 | 1527646590 | 183181009
2018/05/30/10:16:28.191790 | 76083 | 1527646590 | 195183812
2018/05/30/10:16:28.195917 | 76084 | 1527646590 | 210181008
2018/05/30/10:16:28.205225 | 76085 | 1527646590 | 222181009
2018/05/30/10:16:28.228701 | 76086 | 1527646590 | 234183818
2018/05/30/10:16:28.239798 | 76087 | 1527646590 | 249181009
2018/05/30/10:16:28.246009 | 76088 | 1527646590 | 261181009
2018/05/30/10:16:28.269518 | 76089 | 1527646590 | 273183818
2018/05/30/10:16:28.274090 | 76090 | 1527646590 | 288181009
2018/05/30/10:16:28.283431 | 76091 | 1527646590 | 300181008
2018/05/30/10:16:28.307218 | 76092 | 1527646590 | 312183819
2018/05/30/10:16:28.312099 | 76093 | **1527646588 | 312102009**
2018/05/30/10:16:28.323007 | 76094 | 1527646588 | 324102009
2018/05/30/10:16:28.351244 | 76095 | 1527646588 | 336104824
2018/05/30/10:16:28.354883 | 76096 | 1527646588 | 351102008
2018/05/30/10:16:28.362200 | 76097 | 1527646588 | 363102009
2018/05/30/10:16:28.385462 | 76098 | 1527646588 | 375104819
2018/05/30/10:16:28.390903 | 76099 | 1527646588 | 390102009
-----------------------------------------------------
this order is error;
is this error from mpu9250 or opencr-software, maybe blocking??
is there someone help me ? 
thanks!

OpusK, 2018-06-01:
If you do not mind updating, first test it with OpenCR 1.2.0

gavinwangpj, 2018-06-01:
thank you；
ok,next,i will test. 
Can you guess the source of the error

gavinwangpj, 2018-06-05:
after test opencr 1.2.0 and turtlebot3_msgs 1.0.0
It's sometimes ok, and sometimes error.
it's still exist;
I don't know where it from, and also good at opencr;
just test slam by turtlebot3

routiful, 2018-06-07:
I will test it as soon as possible. Sorry for inconvenience.

gavinwangpj, 2018-06-11:
Thanks for the team ——opencr ;)

routiful, 2018-07-12:
Hello @gavinwangpj  :)

We updated new firmware (v1.2.1) and software today
This update includes modified time function in roslib and amended config file for cartographer

After fixed it, the cartographer was not crashed with IMU and without IMU for a long time.

You can get more information about it in [ROS Discourse](https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888/6) 

Thank you for your inquiry and suggestions!

robotpilot, 2018-07-12:
Hi @gavinwangpj
Thank you for your help and supports.
I close this issue. You can reopen this issue to show this issue to the users whenever.
Thanks, :)

closed 2018-07-12 by robotpilot

gavinwangpj, 2018-07-12:
Thanks for your  release!
I will test the new firmware now;