ROBOTIS-GIT/OpenCR #102: About Bootloader

crprobot, 2018-06-09:
Hi, I used "make" to compiled the bootloader file and generate the .bin ,and when i replace the bin file in arduino package ,i flash it successful ,but the bootloader seem not work，the arduino power led was not lighten up,Do you know why? or there is some problem in the code ?

OpusK, 2018-06-11:
What version of the GCC ARM compiler are you using?

crprobot, 2018-06-12:
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)

crprobot, 2018-06-13:
I can compile it now in windows，but I want to know can i access serial port in the bootloader ? I want to download arduino program via serial port.

OpusK, 2018-06-14:
Currently, the bootloader can only support USB. To use the bootloader with Serial Port, you need to add/modify the code.

closed 2018-06-20 by robotpilot