ROBOTIS-GIT/OpenCR #103: Using USBSerial and Serial1

bberrevoets, 2018-06-12:
Hello,
When i want to use the USBSerial and Serial1 at the same time with different speeds, then suddenly my computer is not seeing the board any more, and i get i dmesg the following.

`[10609.183641] cdc_acm 1-1.2:1.0: failed to set dtr/rts`

with `ls /dev`, i also cannot see the port `/dev/ttyACM0`.

When i set both ports to 19200, then everything is working.
Can you explain/fix this problem, because the hardware connected to Serial1 can only work with 19200, and i want the USB connection at 115200.

Kind regards,
Bert Berrevoets.

OpusK, 2018-06-15:
I made simple code for USB and Serial1 (0,1) test.
I checked that this code works normally.

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial1.begin(19200);
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  static bool led_state;
  static uint32_t t_time = 0;

  if(Serial.available())
  {
    Serial1.print(Serial.read());
  }

  if(Serial1.available())
  {
    Serial.print(Serial1.read());
  }

  if(millis() - t_time > 500)
  {
    t_time = millis();
    digitalWrite(LED_BUILTIN, led_state);
    led_state = !led_state;
  }

}
```

I have provided the code to show that each port is independent.
Would you test it out?

bberrevoets, 2018-06-15:
Hello,
First i tried your program, it does not work, then i did a burn a new bootloader on the board, then your program was working, i did program my own program, that was not working. But if i do the bootloader again then my program, it is working. So each time i reload the bootloader, the first program is working, a second program is not working anymore, even if this is the same program you reload on the board.

OpusK, 2018-06-18:
Hi, @bberrevoets 

Is it uploaded to the current board, but only program operation is different?
Does all the other programs show the same thing? Otherwise, is that only the sketches related to Serial, USB?

bberrevoets, 2018-06-24:
Hi OpusK,
No, there is no problem when I do not use Serial1. At the moment I use Serial1 in a uploaded program, I have to rewrite the bootloader, because the only way to do something with the board, is to start it as a DFU to burn the bootloader.

bberrevoets, 2018-06-25:
Hi @OpusK ,
Strange, i just removed the board package from Arduino IDE, reinstalled it, burned the bootloader, and suddenly all my problems where gone. Strange but ok it is working now.

closed 2018-06-25 by bberrevoets