ROBOTIS-GIT/OpenCR #105: Maybe update to C++ standard?

KurtE, 2018-06-20:
I have noticed that several of the other Arduino platforms have updated to more recent C and C++ standards in their builds.

Examples: Teensy for C++ has -std=gnu++14 

I believe that Due has: C++ as -std=gnu++11 C as gnu11

ESP32 - C gnu99   C++ gnu__11
ESP8266 C gnu99 C++ c++11

One reason I am suggesting this, is I am finding code that I compile for some platforms, fails to compile on the OpenCR board.  

Some are simple like:  uint8_t *buffer_ptr = nullptr;

Some are more complex, like using constexpr on class constructors, which allow for them to be initialized...

Beyond simply changing platform.txt to add the the -std= value like: -std=gnu++14

I am not sure what else you might run into.  I know that it will point out code that needs to be changed either with new warning or error messages. 

Example in the turtlebot3 code:  There is:

    static int32_t last_tick[WHEEL_NUM] = {0.0, 0.0};

Which will give a compiler error as you are trying to initialize an integer with a floating point number.

KurtE, 2018-06-20:
Thought I would mention that I have tried compiling turtlebot3 waffle with gnu++14, 
I had to fix two of the above conversions... 

I have not tried running it.  Also it is showing several compiler warnings, which are probably good things to fix regardless if you wish to update the compiler setting.  There are several functions in the DynamixelWorkbench which reach the end without returning something which gives a warning. 

Example:
```
int32_t DynamixelWorkbench::itemRead(uint8_t id, const char* item_name)
{
  static int32_t data = 0;

  if (driver_.readRegister(id, item_name, &data))
    return data;
}
```
What should be returned if readRegister returns 0?

routiful, 2018-06-21:
Hi @KurtE :)

Thanks for sharing it. I have a plan to update Dynamixel Workbench near future. 
I will check all warnings in it.

Thanks
Darby

closed 2018-07-02 by OpusK