ROBOTIS-GIT/OpenCR #108: Compile failed with "cmd_read_board_name fail :0xF020" error

Kanelankai, 2018-06-25:
I met with an error when I want to write turtlebot3_core to OpenCR.
Do you have any idea about this error?
Thank you very much!

OpusK, 2018-06-25:
This error occurs when bootloader problem is occurred.

Did you have tried enter bootloader manually? (please refer to [this link](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#firmware-recovery-mode))

Kanelankai, 2018-06-26:
@OpusK Thank you very much! That solved my problem! Maybe I haven't uploaded turtlebot3_core completely last time.
Thank you!

Kanelankai, 2018-06-26:
@OpusK Sorry for disturbing you once again.
I have succeeded in uploading turtlebot3_core under the firmware recovery mode.
However, after uploading it, PC could not detection ttyACM0 anymore, and the USER3 LED is on with a red light. Do you know the reason about it?
Thank you.

closed 2018-06-26 by OpusK

Wolfman038, 2020-07-30:
ok so im having the exact same problem, but what do i do after i enter the bootloader?

ROBOTIS-Will, 2020-07-30:
@Wolfman038 After entering the bootloader, you can try downloading a firmware or Arduino sketch on OpenCR.

Wolfman038, 2020-07-30:
thank you will try this!

Wolfman038, 2020-07-30:
ok so i tried it on my laptop instead of my desktop which makes me think its an issue with my PC.

ROBOTIS-Will, 2020-07-31:
Thank you for the updates. Please let us know if you have additional inquiry.