ROBOTIS-GIT/OpenCR #112: Dynamixel to Extended Position Control Mode

mpena2099, 2018-07-05:
Hi!

I have an XL430-W250-T and I'm trying to set operating mode (http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#operating-mode11) to "4" which will setup the motor to "extended position control" mode(AKA Multi-turn Mode). This lib support it or just mode 1 (velocity) and 3 (position)?

If this mode is not supported at the current time, could someone point me in the direction of how to do this?

Thanks!
Mauricio

mpena2099, 2018-07-06:
Hi @routiful!

What I'm trying to do is something like this, using workbench:

```
dxl_wb.jointMode(DXL_ID);
dxl_wb.itemWrite(DXL_ID, "Operating_Mode", 4);  //  multi-turn mode
dxl_wb.goalPosition(DXL_ID, 0);  

```

Mode 4 do not accept values above 4095...

If I remove de first of the 3 lives above, goalPosition do not work.

Thanks!
Mauricio

mpena2099, 2018-07-06:
Hi all. 

I got the answer: CURRENT MODE, as seen in https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/08.%20DynamixelWorkbench/n_Current/n_Current.ino

I have no idea what is the difference between "Extended Position Control Mode" and "Current-based Position Control Mode". The first is mode 4 and the second mode 5.

Look for X_SERIES_POSITION_CONTROL_MODE  and X_SERIES_CURRENT_BASED_POSITION_CONTROL_MODE  in https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/DynamixelWorkbench/include/dynamixel_workbench_toolbox/dynamixel_workbench.h 
Thanks!

KurtE, 2018-07-06:
As I mentioned on forum: dxl_wb.itemWrite(DXL_ID, "Operating_Mode", 4);
I think will not work as torque is on.

Maybe something like

dxl_wb.itemWrite(DXL_ID,  “Torque_Enable“, 0);
dxl_wb.itemWrite(DXL_ID, "Operating_Mode”, 4);
dxl_wb.itemWrite(DXL_ID, "Torque_Enable", 1);

ROBOTIS-Will, 2018-07-06:
@mpena2099 
Please avoid posting the identical question on multiple sites because it makes us difficult to provide seamless support to you.
[http://en.robotis.com/service/forum_view.php?bbs_no=2406887&page=1&save_sca=&sca=&save_stx=&stx=&sfl=](http://en.robotis.com/service/forum_view.php?bbs_no=2406887&page=1&save_sca=&sca=&save_stx=&stx=&sfl=)
Above forum thread will note be updated as @routiful will help you in this page :)
Thank you.

routiful, 2018-07-06:
As @KurtE commented, JointMode() includes `torque on` commands. (Thnaks @KurtE !)
So you are supposed to `torque off` before set `Operating_Mode`.

Becasue XL series doesn't support `CURRENT_BASED_POSITION_CONTROL_MODE` (http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#operating-mode11), user can't set it in XL series.
I recommend to use `EXTENDED_POSITION_CONTROL_MODE` instead of it.

The difference of between `CURRENT_BASED_POSITION_CONTROL_MODE` and `EXTENDED_POSITION_CONTROL_MODE` is to use different control input.
`CURRENT_BASED_POSITION_CONTROL_MODE` uses `current` but `EXTENDED_POSITION_CONTROL_MODE` uses `voltage` to control position of actuator.

mpena2099, 2018-07-09:
@ROBOTIS-Will , thanks for the advice.

@KurtE code worked! Why current mode worked for me? Because previous attempts to change the operation mode to 4 saved it on motor... :) (And trying to change to mode 5 do nothing)

Thanks!
Mauricio

closed 2018-07-09 by mpena2099

routiful, 2018-07-09:
I'm glad to solve your issue. If you have any questions or issues, please feel free to get new ticket :)