ROBOTIS-GIT/OpenCR #113: develop branch compile issue.

KurtE, 2018-07-09:
After the merge of my Wire updates was completed, I synced up my develop branch and was going to try out the new I2C timings.

When I tried to compile scanner.ino, the compile failed:
```
"C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2/bin/arm-none-eabi-g++" -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10805 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR     -mthumb  -DSTM32F746xx -D__OPENCR__ "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/bsp/opencr" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/bsp/opencr/include" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/hw" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/hw/driver" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/hw/usb_cdc" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/STM32F7xx_HAL_Driver/Inc" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps/client/include" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps/client/include/micrortps/client" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps/client/src/c" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps/cdr/include" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps/transport/include" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps/transport/include/micrortps/transport" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR/lib/micrortps/transport/src/c" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\cores\arduino" "-IC:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR" "C:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR\lib\micrortps\micrortps.cpp" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_216550\core\lib\micrortps\micrortps.cpp.o"
In file included from C:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR\lib\micrortps\micrortps.cpp:11:0:

C:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\variants\OpenCR\lib\micrortps\micrortps.hpp:12:31: fatal error: microcdr/microcdr.h: No such file or directory

compilation terminated.

Using library Wire in folder: C:\Users\kurte\Documents\Arduino\hardware\opencr\opencr\libraries\Wire (legacy)
exit status 1
Error compiling for board OpenCR Board.
```
Looks like a new variant library was added 12 days ago: opencr\variants\OpenCR\lib\micrortps
and there is no sub directory microcdr... Not sure if missing or supposed to bring in from somewhere else?

OpusK, 2018-07-10:
Oh, I'm sorry.
It is my mistake.

Problems occurred while merging other branches.
I will fix it quickly.

OpusK, 2018-07-10:
Thanks @KurtE.

I restored it to its original state.
It was my mistake that I was wrongly merged with the branch related to ROS 2.0.

Thanks to your issue, I was able to process it quickly. thank you :)

closed 2018-07-10 by OpusK