ROBOTIS-GIT/OpenCR #114: Windows not recognizing OpenCR board

eng-Sund, 2018-07-09:
Hi,

I am getting a “Failed to open port” error when I try to upload any code. Previously when I was getting this error I disconnected the Dynamixel while uploading and it seemed to have fixed the error. But, now I am getting the error once again and windows does not recognize the board. I tried common troubleshooting techniques such as restarting the PC, reinstalling the driver, trying a different PC etc. to no avail.

I tried to follow the firmware recovery process mentioned in the manual but I am not even able to open the opencr_ld.exe file. A window pops up and it immediately closes. Also, I am unable to install a new driver using zadig as windows does not recognize the device. 

Board: Open CR 1.0 
OS: Windows 10

Best,
Prasanna

OpusK, 2018-07-10:
Hi, @eng-Sund 

When using DFU mode, is the device recognized by Device Manager as below?
[image]

opencr_ld.exe should not be just open, but [should be commanded with parameters in cmd.](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#windows-example)

eng-Sund, 2018-07-10:
Thanks for the quick reply.

I entered DFU mode and updated the driver using zadig.  This is what shows up in device manager. Before updating driver, there was a yellow exclamation beside "STM32 BOOTLOADER"

[image]

Now, that my driver is installed, I reset (and press SW2) the device and windows did not recognize the device again, but it does in DFU mode. **But, I am able to enter firmware recovery mode so I guess the firmware is corrupt?**

Also, I figured I will have to use the command prompt for the opencr_Id.exe file but didn't know what/where the opencrfw.bin file was. **How do I generate the bin file?**

Finally, two more questions:

**1. What is the difference between using mavlink and the firmware recovery mode?
2. When do I know if I need to update the bootloader?**

I apologize my lack of knowledge and appreciate your patience

OpusK, 2018-07-11:
If you want to re-burn the boot loader, use ["Burn Bootloader"](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#download-bootloader) features in Arduino.

> But, I am able to enter firmware recovery mode so I guess the firmware is corrupt?

It is rare that the boot loader firmware is corrupted, but if it does, you can not enter recovery mode.

> How do I generate the bin file?

Using the [Sketch] -> [Export compiled Binary] function in the IDE's IDE menu creates a binary file in the sketch folder.

> What is the difference between using mavlink and the firmware recovery mode?

Firmware recovery mode is a function that enables you to upload application firmware by manually entering the boot loader, and the communication method used by the boot loader is mavlink.

> When do I know if I need to update the bootloader?

The boot loader has not been updated yet.
If your question is asking when to re-burn the boot loader, you can do so if the USB driver is not found in the operating system in firmware recovery mode. (Used as a last resort if it is not possible through various attempts such as driver installation)

eng-Sund, 2018-07-11:
Hi,
Thanks for the reply! I was able to burn the bootloader in DFU mode. So, I think, it is the firmware (assuming only two things can go bad: bootloader or firmware as windows still doesnt recognize the board) that is corrupt. I am trying to reinstall (if that is the right word) the firmware or update the firmware. But I did not understand:

1. After generating the communication codes using mavelink, how do I run commands for the boot loader to download and execute the firmware? Command prompt? Visual Studio?

2. What is the folder structure used for? http://emanual.robotis.com/docs/en/parts/controller/opencr10/#folder-structure

3. Process of creating the binary file: I do get the steps but which file am I using to create the file. Can it be any code/ino file? In the example, in one image it says "turtlebot3_core", in one it says "opencrfw.bin" and in the last it says "Blink.ino.bin"

4. Finally, is there a difference between running the commands in step and using the binary file in step 3 for updating firmware? That was my question earlier, but I think it might have been misunderstood.

Also, the xml link provided doesn't work, as it has not been updated. The new link is:
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_develop/opencr_bootloader/common/msg/xml/opencr_msg

OpusK, 2018-07-12:
1. Is there any reason to use Marvink? If you have already booted the boot loader in DFU mode, this step is not necessary. If you want to modify or create a new boot loader, it is correct to see the manual contents.

2. The folder structure is the folder structure of the [bootloader](https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_develop/opencr_bootloader).

3. Typically, this is "sketch_name.ino.bin". In your example it will be turtlebot3_core.ino.bin.

4. There is no difference. When uploading from Aduno, the command is executed by automatic script.

In my opinion, this is not a boot loader problem, but a USB port recognition problem.
If you are using a USB hub, do not use it, but connect directly to your PC.
And, manually enter the boot loader and check that the [STATUS LED](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#user-led) blinks.

eng-Sund, 2018-07-12:
Thanks once again for the quick reply. So, I was able to solve the issue by downloading the firmware in DFU mode as mentioned [here](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#firmware-recovery-mode).

I opened the turtlebot3_core.ino and created a binary. Then, I renamed it to opencrfw.bin and followed the steps mentioned in the website.

Without that, the board was being recognized in DFU mode and when I "executed" the bootloader. But, once I got out of it (hit the rest button), windows stopped recognizing the board. I was always connected to the USB port of the pc and did not use a USB hub. Thanks for the help. 

I just had 2 final questions before closing this thread:

1. Instead of turtlebot3_core, could I have used any file to create the binary? Or is there something special about that ino file? For example, could I have used blink.ino?

2. How do I prevent this problem of the board being not recognized by windows from re-occuring? Is the issue random or did I do something wrong? I would prefer if I don't have to re-download the firmware each and everytime it happens.

Thanks

OpusK, 2018-07-13:
1. That's enough. I think it would be nice to use the lightest firmware for these tests. It is recommended to test it with an example like Blink.
<br>

2. There are a variety of reasons for this, so sometimes the user may prevent it, but it may not prevent it.
 - Normally, if the board is stuck because the firmware is wrong, USB recognition is not possible, so you need to upload the normal firmware.
 - Due to the nature of USB, OS problems may cause recognition problems. In this case, you need to react in terms of OS such as rebooting or rebooting USB connection.

First of all, as I said in 1, it is recommended to test from simple firmware.

closed 2018-07-13 by eng-Sund

tomsepe, 2019-03-21:
I was having a similar issue with Windows 10 and the OpenCR1.0 board. I could only upload firmware by pressing the SW2 and reset buttons. But then the board would still not show up as a USB device so I could not communicate with it over the USB/Serial COM port.

I resolved the issue by using this driver fixer and now I can upload firmware to the board from windows 10 via the Arduino IDE, and I can open a serial monitor and communicate with the board without having to fiddle with tiny bottons.

impulseRC driver fixer
https://impulserc.blob.core.windows.net/utilities/ImpulseRC_Driver_Fixer.exe
https://www.dronetrest.com/t/fix-any-stm32-dfu-drivers-issues-when-flashing-betaflight-cleanflight-firmware/3603

ottensmeyer, 2021-01-13:
Hi tomsepe and all - 

tomsepe - could you provide additional details on how you solved your problem?

I also have an OpenCR that used to be recognized by Win10, showed up as serial port COM6, has just stopped being recognized.  Rebooting Windows didn't help, Zadig says it was successful in installing the driver (at it shows up correctly when I check the driver when in DFU mode, but when I power-cycle the OpenCR1.0 it doesn't show up in the list of ports or as a USB device.  

When in bootloader mode (SW2), I can upload my code from the Arduino IDE, but as soon as it finishes, the COM6 (in my case) entry in the device manager disappears, and I can't open the serial monitor.

I downloaded impulseRC, put the OpenCR into DFU mode (SW0), impulseRC found and replaced the driver, but the situation has not changed from the behavior described above.  Tried Zadig again, still no joy.

I can reach DFU mode, I can reach bootloader mode, can upload firmware, but I need access to the serial monitor (or to establish a serial connection between my laptop and the OpenCR to provide commands from a separate application) to test and interact with it.  

Is there any way to set the OpenCR so that it defaults to bootloader mode?

tomsepe, 2021-01-13:
@ottensmeyer All I did was run the impulseRC driver fixer and that solved the problem for me.

ottensmeyer, 2021-01-13:
Thanks for getting back to me - will have to keep looking - going to see if it talks to my linux machine.

ROBOTIS-Will, 2021-01-15:
@ottensmeyer 
If your Windows 10 is connected to internet, Windows will install proper USB drive automatically.
Please try uninstalling the current driver as below when selecting `Uninstall Device` from USB properties window.
[image]
Disconnect the OpenCR and reconnect again to trigger the Windows driver installation.

ottensmeyer, 2021-01-15:
Thanks all - turns out it was a software issue - incorrectly converting dynamixel example code for Arduino from the "everything-in-setup()" form to one where the dynamixel elements are global and accessed in loop() and other functions.