ROBOTIS-GIT/OpenCR #117: No Memory Allocation in Dynamixel Read Position/Velocity functions for protocol 1.0

erikgroving, 2018-07-19:
In open_manipulator_motor_driver.cpp located here:  **OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/open_manipulator/open_manipulator_motor_driver.cpp**:
If the protocol is not version 2.0, no memory is allocated to the pointer, and thus invalid memory accesses occur in the following for loops at lines 133 and 161.

See the following:
```
  int32_t* get_joint_present_position = NULL;

  if (protocol_version_ == 2.0)
    get_joint_present_position = joint_controller_.syncRead("Present_Position");
  else if (protocol_version_ == 1.0)
  {
    for (int index = 0; index < JOINT_NUM; index++)
      get_joint_present_position[index] = joint_controller_.itemRead(dxl_id_[index], "Present_Position");
  }
```
In the case the the protocol version is 1.0, get_joint_present_position should be allocated memory. The memory can then be freed after the values are set to **present_position**:
```
  for (int index = 0; index < JOINT_NUM; index++)
    present_position[index] = get_joint_present_position[index];

```

The same occurs for `readVelocity` function. I have made these changes on my local machine, if you would like me to create a pull request, I could do so.

routiful, 2018-07-23:
Hi :)

Thank you for your contribution! I didn't recognize that problem. That is my mistake.
I will update it as soon as possible. 

Thank you again

erikgroving, 2018-07-23:
Great thanks :)

closed 2018-07-23 by erikgroving