ROBOTIS-GIT/OpenCR #118: Unable to control 5 joints, max seems to be set to 4

erikgroving, 2018-07-20:
Hi,
I am using the Dynamixel workbench to control a custom manipulator using ROBOTIS-supported Dynamixels via turtlebot. I have modified the firmware to support the arm. It seems however, that 5 joints are not supported at the moment, I know that the open manipulator has only 4 joints so perhaps more than 4 joints was not considered. When I have 5 joints, this function never returns: `joint_controller_.syncWrite("Goal_Position", goal_position);` and the robot arm does not move at all as seen [here](https://ibb.co/medAjJ)

However, when I set JOINT_NUM to 4, everything runs perfectly, however, this is not a viable alternative because it means one joint is just hanging loosely.  The result can be seen[ here](https://ibb.co/kjmVjJ), with the 5th joint loose.

Do you have any idea why this could be? I am using the dynamixel workbench. Could it be related to these defines in **dynamixel_driver.h**?
```
#define MAX_DXL_SERIES_NUM 3
#define MAX_HANDLER_NUM 5
```

Thanks.

erikgroving, 2018-07-20:
Okay, I think I have something that works but I have no idea why. I had originally done this:
```
#define JOINT1    11
#define JOINT2    12
#define JOINT3    13
#define JOINT4    14
#define JOINT5    15
#define GRIPPER   16
```
Using the Serial Debug on UART2 and the OpenCR library, I observe the devices being set to joint mode and the length of sync write handlers:
```
Success to init Motor Driver
Device 11 : XL430-W250
Device 12 : XM430-W350
Device 13 : XM430-W350
Device 14 : XL430-W250
Device 15 : YM43YM43YM4350
Length: 32
Added a sync write handler.
SWH cnt: 78
```

As it can be seen, the last dynamixel is a lot of gibberish, and it appears to write past its allocated memory and affect other variables, which resulted in a crash. To test if it was the amount of joints or improper hardware, I switched JOINT4 with ID 14 to 15, and JOINT5 with ID 15 to 14 as such:
```
#define JOINT4    15
#define JOINT5    14
```
To my surprise, the output was as follows:
 ```
Success to init Motor Driver
Device 11 : XL430-W250
Device 12 : XM430-W350
Device 13 : XM430-W350
Device 15 : XM430-W350
Device 14 : XL430-W250
Length: 4
Added a sync write handler.
SWH cnt: 1
```

And the robot arm responded [perfectly](https://ibb.co/nsnUdd), with all joints assuming the start position. It seems there are no invalid memory writes and all IDs are properly parsed. However, all I changed was the order in which the IDs of my joints were being scanned. Furthermore, when I run the find_Dynamixel example, the joints are read in order (11-15) with no problem. For now, I will just leave it as is since it seems to work, but it is a bit concerning as to why this change allowed my hardware to work! Do you have any insights?

erikgroving, 2018-07-20:
I am able to publish to the topic created in the roscore from a remote PC and the arm moves accordingly. The only reason I have not closed the issue is because I am not sure why switching JOINT4 and JOINT5 IDs results in the arm working. It is a rather odd issue/bug.

I get these errors during bringup but I think it is just due to the changes to the firmware that needed to be made to support the arm?

```
[ERROR] [1532071457.184758]: Parameter ~tf_prefix does not exist
[WARN] [1532071458.178027]: Failed to get param: timeout expired
[ WARN] [1532071461.249601604]: Check OpenCR update and change your firmware!!
```

routiful, 2018-07-23:
Hi :)

Sorry for late reply and Thank you for your inquiry about OpenManipulator.
To use open_manipulator_motor_driver in turtlebot3 with custom robot arm, you are supposed to modify some parameters

First, The number of Dynamixels including joint and gripper

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/open_manipulator/open_manipulator_motor_driver.h#L24

Second, The number of Joint

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/open_manipulator/open_manipulator_motor_driver.h#L25

Third, Dynamixel ID (related Joint)

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/open_manipulator/open_manipulator_motor_driver.h#L33

Fourth, Initialize `dxl_id_` array

 https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/open_manipulator/open_manipulator_motor_driver.cpp#L23

If you complete above instructions, you can use your robot arm.

The last question is able to be solved update to latest software.
I attached link to you 

https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888/5

erikgroving, 2018-07-23:
Hi, I had actually done all those steps. However, the issue ended up being much weirder:
```
#define JOINT1    11
#define JOINT2    12
#define JOINT3    13
#define JOINT4    14
#define JOINT5    15
#define GRIPPER   16
```
When I had this, it seems like it would have a buffer overflow and fail. Then when I switched the IDs of JOINT4 and JOINT5 everything worked. I am currently able to control the arm as I want with the IDs switched. I will update the firmware and software and see if that was the underlying cause.

closed 2018-07-23 by erikgroving

routiful, 2018-07-24:
I will check it before next update. I might be miss something. 
Thank you for your report!