ROBOTIS-GIT/OpenCR #121: No longer able to connect to dynamixels after modifying core firmware

erikgroving, 2018-08-01:
Hi,
I modified the firmware on the turtlebot 3 with open manipulator to allow myself to control the velocity of joints by setting up a subscriber on a new topic such that when it would receive a message, it would run the following code:

```
for (int i = 0; i < JOINT_NUM; i++) {
    joint_controller_.jointMode(dxl_id_[i], value[i]);
}
```
It initially worked when I was manually published to the topic, however, when I ran my program I lost all contact with my Dynamixels. The Dynamixels do show a red light when the power goes on but find_dynamixel and R+ manager and the single_manager in Dynamixel Workbench all are unable to find any Dynamixels.

I am using both the XM430-W210 and XL430-W250. I used the usb_to_dxl image. 

Do you have any suggestions? Even when connecting only 1 dynamixel to the opencr board, I still do not successfully scan anything.

routiful, 2018-08-02:
Hi :)

I don't understand your development environments. Can you explain how to set it?
And how do you want to control Dynamixels? OpenCR or PC?

erikgroving, 2018-08-02:
Hi,
The firmware was modified to add a goal_joint_velocity subscriber and callback. The callback uses the joint_controller_ from OpenManipulator.h to run the above for loop that changes the velocities of each dynamixel.

I interfaced my arm with MoveIt, and would like to control the velocities as a result. I successfully was able to move with moveit but the arm moved to fast. When I uploaded the firmware, it worked initially, however it stopped working after a bit. Could I have mistakenly written erroneous data to the registers in the Dynamixels that made them stop responding? I have only used the jointMode function which seems to just write to a register on the Dynamixel, so I don't know why the Dynamixels would have stopped responding. There are 5 dynamixels connected in a link, containing the 2 models above.

routiful, 2018-08-02:
MoveIt can makes profile from point to point, I used to it before. I think this is more simple solution.
I attached below link. This issue includes a comment how to make smoothing motion in MoveIt

https://github.com/ROBOTIS-GIT/open_manipulator/issues/10

and below link might be help you too.

http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/time_parameterization/time_parameterization_tutorial.html

erikgroving, 2018-08-02:
Hi! The problem is not with MoveIt, I have already configured MoveIt and it was working like a charm before the Dynamixels failed to be pinged.

The problem is that currently all the Dynamixels are not responding. When I turn on the turtlebot, the red indicator light will flash on the Dynamixels, however, the ping to the Dynamixels all fail. My question is about why the Dynamixels would not longer be ping-able and what I could do to fix this.

routiful, 2018-08-03:
I don't understand.
We already provide information for joint position and velocity by `/joint_states` topic.
If you uploaded firmware(turtlebot3_with_open_manipulator_core.ino), you can see this topic.

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_with_open_manipulator/turtlebot3_with_open_manipulator_core/turtlebot3_with_open_manipulator_core.ino#L447

And you can control the dynamixels by using `open_manipulator_with_tb3/goal_joint_position` topic

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_with_open_manipulator/turtlebot3_with_open_manipulator_core/turtlebot3_with_open_manipulator_core_config.h#L141

I think you don't need to any amend the firmware, if you want to get a velocity of joint.

erikgroving, 2018-08-03:
Hi! Actually the issue is not with velocity, I was just providing context.

The issue is when the OpenCR board is turned on, the SERIAL_DEBUG output prints 'Failed to ping' for the Dynamixels for my arm. The find_dynamixel example also fails to find them. The dynamixel's will still flash red when turned on though

erikgroving, 2018-08-05:
Here is a previous output from T_findDynamixel:
```
9600 bps
57600 bps
1000000 bps

id : 1   Model Name : XL430-W250

id : 2   Model Name : XL430-W250

id : 11   Model Name : XL430-W250

id : 12   Model Name : XM430-W350

id : 13   Model Name : XM430-W350

id : 14   Model Name : XL430-W250

id : 15   Model Name : ZM430-W350

id : 16   Model Name : XL430-W250

End
```

Currently the output looks like this:
```
9600 bps
57600 bps
1000000 bps
End
```

My guess is that the controller or EEPROM inside the Dynamixels has been affected. Is there anyway to reflash the controller inside the Dynamixel?

routiful, 2018-08-06:
You can use `firmware recovery` to Dynamixel by using R+ manager 2.0 

http://emanual.robotis.com/docs/en/software/rplus2/manager/#firmware-recovery

erikgroving, 2018-08-06:
[This popup appeared as well](https://image.ibb.co/cvMuTz/image.png)

Does this mean that for the Dynamixels of series XL that are not XL320, I must use the USB2Dynamixel special adapter? [This one](https://www.trossenrobotics.com/robotis-bioloid-usb2dynamixel.aspx)

routiful, 2018-08-07:
I tested XL430-W250-T and sud_to_Dxl firmware. But it is working well.
Please check **12v power** through OpenCR and you need to **turn off and on** the OpenCR before recovery.

And your last commented is not related this issue.

erikgroving, 2018-08-07:
Hi,
The power is coming from a 12 volt adapter that was working previously so I do not think that is the issue. I checked the voltage on the white connector next to the power in (pins labelled POWER OUT + - ), and I read 12 volts. I turned the OpenCR board on and off and tried again but it is still not working. Previously, I was able to see all the Dynamixels, but now I cannot even see the Dynamixels used for the turtlebot wheels.

Do you think it could be something with the controller inside the Dynamixel that has gone awry?

erikgroving, 2018-08-07:
When the R+ Manager is running and searching for Dynamixels, this [TX light ](https://preview.ibb.co/e9NAmK/image.png)continuously flashes. No response though. Even though when plugged in, the Dynamixel's red light will turn on for a second

routiful, 2018-08-08:
Hi
Sorry for inconvenience. I think the Dynamixel has faults.
If you send to customer team(contactus2@robotis.com), we would fix it and send back to you.

erikgroving, 2018-08-15:
Just an update to anyone else who encounters the same issue. The microcontrollers in the Dynamixels became faulty due to the use of the joint_mode function defined [here](https://github.com/ROBOTIS-GIT/OpenCR/blob/6cdf5a1a19349032e7b612a19b4bfb784ee97a89/arduino/opencr_arduino/opencr/libraries/DynamixelWorkbench/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp#L185
)

This function only seems to modify the Profile Acceleration and Profile Velocity registers, however, perhaps writing to this function too fast may result in faulting the Dynamixel. I am currently in the RMA process for the Dynamixels.

Should I close the issue?

routiful, 2018-08-16:
JointMode function make set position control mode and parameter for making profile(velocity and acceleration).
I don't think this function has problems. It calls only one before Dynamixel is enabled(torque_enable).

erikgroving, 2018-08-16:
Would it be bad to use it more than once? And in rapid succession? If that is the case, that could be why the Dynamixels became faulty.

routiful, 2018-08-16:
Would you upload an example that makes faulty and explain to me your development environments? 
It might be helpful to me to check it and update JointMode functions.

erikgroving, 2018-08-16:
Okay! By the way, I just want to say thank you so much for taking a look into this, I really appreciate it. Basically, the arm was moving too fast, so I wanted to control velocity. 

I added a subscriber to a new topic in turtlebot3_config.h:
`ros::Subscriber<sensor_msgs::JointState> joint_vel_sub("grasslab_manipulator_with_tb3/goal_joint_velocity", goalJointVelocityCallback);`

I set up the subscriber in the `setup()`:
`nh.subscribe(joint_vel_sub);`

The callback is as follows:
```
bool GrassLabManipulatorMotorDriver::writeJointVelocity(uint16_t* value) {
  for (int num = 0; num < JOINT_NUM; num++) {
    joint_controller_.jointMode(dxl_id_[num], value[num]);
  }
  return true;
}
```

The GrassLabManipulatorMotorDriver class is pretty much the same as OpenManipulatorMotorDriver, except  number of joints and the IDs of the joints are different to fit our custom arm.

When MoveIt computes a plan, it sends a FollowJointTrajectory action to a controller. This has a vector of positions, velocities, efforts, etc... I decided to publish the velocities generated from MoveIt to the topic with the `goalJointVelocityCallback`. Since a plan generated by MoveIt can have many points, it is possible that the callback was called several times quite quickly.

I tried to be as clear as possible, but if there's anything that's not clear, let me know.

routiful, 2018-08-17:
If you want to change joint velocity, you don't need to call jointMode function every time.
This function includes complex commands to access torque, operating_mode register.
I think you only need to change velocity of Dynamixels. I attached some codes

```
bool GrassLabManipulatorMotorDriver::writeJointVelocity(uint16_t* value) {
  for (int num = 0; num < JOINT_NUM; num++) {
    // joint_controller_.jointMode(dxl_id_[num], value[num]);
    joint_controller_.syncWrite("Profile_Velocity", (int32_t *)(value)); // write velocity
  }
  return true;
}
```

erikgroving, 2018-08-17:
Thank you so much! I will change my writeJointVelocity function to be that.