ROBOTIS-GIT/OpenCR #123: error: Connection refused

wsung1, 2018-08-23:
Hi, ROBOTIS,
While following the textbook - ROS robot programming, I got stuck in 9.2.6. rosserial examples (p.276). As written in the textbook, I first downloaded a_LED.ino to the OpenCR.
[image]

Then, I tried to run rosserial_server using rosserial_python as shown below but it failed with the following error:
[image]

Please help me solve this problem.
Thanks in advance :)

kijongGil, 2018-08-24:
Hi wsung1 :)
I think ros is installed incorrectly, or the python configuration seems to be wrong.
could you try to install ros-serial?
`sudo apt-get install ros-kinetic-rosserial ros-kinetic-rosserial-server ros-kinetic-rosserial-arduino
`

wsung1, 2018-08-29:
Thanks for your reply @kijongGil :) 
The cause was turned out to be activating ROS_MASTER_URI.
https://answers.ros.org/question/248280/rosserial-arduino-connectivity-problem-socketerror-errno-111-connection-refused/

closed 2018-08-29 by wsung1