ROBOTIS-GIT/OpenCR #124: Searching for API document

wsung1, 2018-08-29:
Hi, @OpusK :)
I'm walking through the CAN examples you made such as canTxRxMessage.
It says /* For API doc, please refer to https://github.com/ROBOTIS-GIT/OpenCR/wiki/arduino_examples_can */ but nothing could be found via this link.
Where can I obtain the API document?
Thank you, @OpusK!

wsung1, 2018-09-16:
Thanks for replying @OpusK.
So, is there no way to approach the API document right away?

OpusK, 2018-11-08:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-08 by OpusK

amburkoff, 2018-11-13:
please do emanual and usage examples

OpusK, 2018-11-13:
@amburkoff 

[APIs and example of CAN have been updated today](https://github.com/ROBOTIS-GIT/OpenCR/commit/11bba7a3c5a526664a9eb679b2fb25a2d66ac1c8). 
These are currently in develop branch, and we will add it to emanual when it is officially released.