ROBOTIS-GIT/OpenCR #125: CAN port in OpenCR

wsung1, 2018-08-29:
Hi, ROBOTIS,
Among several communication ports that serve in OpenCR, only CAN has no information except its connector is 20010WS-04 (4-pin male connector).
I'm also wondering why there are 4 pins for the one single CAN channel.
Please let me know at least the pin map for this in that which one is CAN high/low and so on.
Thanks in advance!

OpusK, 2018-09-15:
Hi, @wsung1 ,

Sorry for late reply you.
Please refer to the figure below. (This is also shown on the OpenCR board)
[image]

The CAN pin map will be updated later in e-manual.

wsung1, 2018-09-16:
Thanks for replying @OpusK, but I'm still wondering what G and V pins do.
As far as I know, CAN generally consistas of H and L.

OpusK, 2018-09-17:
As you say, CAN can communicate using High / Low only.
It seems to have been added at the hardware design stage because it may require G / V depending on the user.

KurtE, 2018-09-18:
@wsung1, obviously someone from Robotis can answer this better than I can... 

But for issues like, this, I usually find it helps to look at the schematic and other design files. 
You can find the Schematic through the E-Manual or directly from github: https://github.com/ROBOTIS-GIT/OpenCR-Hardware/blob/master/Schematic/OpenCR_REVF_Schematic.pdf

So if you look at Sheet 8 at the CAN section you will see that:
G - Ground
V - Is voltage of what looks like 5 Volts.

So these two pins allow you to apply power and ground, which may be needed by a Can device.  Note: You almost always require a ground...

wsung1, 2018-09-18:
@KurtE, I'd appreciate for your pieces of advice. It helps me a lot! :)

robotpilot, 2018-09-30:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-09-30 by robotpilot

kaif-46, 2022-01-15:
@OpusK @KurtE @wsung1 Will, you be able to provide some information about how to access these pins in the code and how can we send/receive data over it?

KurtE, 2022-01-15:
Sorry I have not played with OpenCR for a couple of years now... So not much help on this one. anymore.

ROBOTIS-Will, 2022-01-17:
@kaif-46 

Please create a new issue thread since this thread had been closed long ago.

You may refer to the CAN example provided in the OpenCR Arduino board manager.
[image]

Please checkout our eManual to find out how to install the required library for OpenCR.
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#arduino-ide

Thanks!

kaif-46, 2022-02-02:
How can i change the frequency to 8MHz for CAN communication.

ROBOTIS-Will, 2022-02-03:
Please refer to the thread #296.