ROBOTIS-GIT/OpenCR #129: Connect XL-320 & XL430 to OpenCR

Purmpol, 2018-09-13:
Hi,
I'm learning ROS and want to make a robot based on DXL devices. I'm going to buy OpenCR and servos, but I need to know something before. This is the first question.

Can we connect both XL-320 and XL-430 to OpenCR?  How?

Thanks and admire all Robotis team.

Purmpol, 2018-09-13:
After more studying, I think it's difficult to control when use mixed models of actuators. Because Control Table of EEPROM Area are different.

Perhaps, I should use only one model.

OpusK, 2018-09-15:
Hi, @Purmpol 

> Can we connect both XL-320 (7.4V) and XL-430 (11.1V) to OpenCR? and How?

XL-320 and XL-430 are not recommended for use together because their recommended operating voltages are different.

> After more studying, I think it's difficult to control when use mixed models of actuators. Because Control Table of EEPROM Area are different.

The XL-320 and XL-430 have different EEPROM addresses, but the X-430 and X-540 series have the same address. (540 has extended functionality, but the rest of the address is the same.)
Of course, even if the XL-320 and XL-430 have different EEPROM addresses, it will not be a problem to operate them for each model.

Purmpol, 2018-09-15:
Your answer is clear. Thank you so much.

I have more 2 questions.
1. I don't see OpenCR board and XL-430 on Mouser's list. Do you plan to sell them on Mouser?
(Because in my country, Mouser provides free shipment, and easier customs clearance)

2. How OpenCR "physically" connect to Intel NUC on OP3?
The picture on e-manual, there's only one USB port for firmware download (not always connect to NUC), right?

[image]

Purmpol, 2018-09-15:
Acknowledge. Thanks again for your help.

closed 2018-09-17 by OpusK