ROBOTIS-GIT/OpenCR #130: Can not compile dxl_monitor using current openCR release

KurtE, 2018-09-17:
Recently I updated to the updated the OpenCR board as recommended by board manager to I believe it is 1.3.2.  

I thought I would test current released code to see how my updated code compares (DEV branch).  

Now the Dynamixel SDK example app dxl_monitor does not build on my windows 10 64 bit machine. 
```
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\tools\\opencr_gcc\\5.4.0-2016q2/bin/arm-none-eabi-gcc" -c -g -O2 -std=gnu99 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10807 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/bsp/opencr" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/bsp/opencr/include" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/hw" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/hw/driver" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/hw/usb_cdc" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\cores\\arduino" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR" "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR\\hw\\driver\\drv_adc.c" -o "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino_build_394211\\core\\hw\\driver\\drv_adc.c.o"
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\tools\\opencr_gcc\\5.4.0-2016q2/bin/arm-none-eabi-gcc" -c -g -O2 -std=gnu99 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10807 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/bsp/opencr" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/bsp/opencr/include" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/hw" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/hw/driver" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/hw/usb_cdc" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/lib/STM32F7xx_HAL_Driver/Inc/" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR/" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\cores\\arduino" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR" "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\OpenCR\\hardware\\OpenCR\\1.3.2\\variants\\OpenCR\\lib\\STM32F7xx_HAL_Driver\\Src\\stm32f7xx_hal.c" -o "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino_build_394211\\core\\lib\\STM32F7xx_HAL_Driver\\Src\\stm32f7xx_hal.c.o"
Using library DynamixelSDK at version 0.0.1 in folder: C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2\libraries\DynamixelSDK 
C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2\variants\OpenCR\lib\micrortps\client\src\c\core\session\output_best_effort_stream.c:1:69: fatal error: micrortps/client/core/session/output_best_effort_stream.h: No such file or directory

compilation terminated.

exit status 1
C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2\variants\OpenCR\lib\micrortps\client\src\c\core\session\stream_storage.c:1:58: fatal error: micrortps/client/core/session/stream_storage.h: No such file or directory

compilation terminated.

C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2\variants\OpenCR\lib\micrortps\client\src\c\core\session\stream_id.c:1:53: fatal error: micrortps/client/core/session/stream_id.h: No such file or directory

compilation terminated.

Error compiling for board OpenCR Board.
```

Looks very familiar, like it might be the same issue as I earlier had on develop branch in issue #113 

Probably ROS2 stuff?

OpusK, 2018-09-17:
Oh, sorry for the inconvenience.
This is my mistake.

As you said, the archive contained ROS2 stuff.

I have fixed the bug, please remove 1.3.2 and reinstall. Thank you.

KurtE, 2018-09-17:
FYI - I have run into issues with Arduino 1.8.7 uninstalling (or choosing different version).  
I submitted issue against Arduino github: https://github.com/arduino/Arduino/issues/8006

Last night I was able to make the current version compile, by deleting the microtps directory from it. 

So I was able to build dxl_monitor.

Note: When I bring up board manager and maybe other places, I see several messages that all look something like:
```
Invalid library found in C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2\libraries\turtlebot3_ros2: no headers files (.h) found in C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2\libraries\turtlebot3_ros2
```

Have not been able to uninstall 1.3.2 yet but will try to reinstall and see if that makes it happier...

KurtE, 2018-09-17:
I tried now to use the Install again after the uninstall sort of failed...

Now when I try to install 1.3.2 it complains and fails with message at bottom saying:
CRC doen't match, file is corrupted.  It may be a temporary problem, please retry later.

And now it is not allowing me to startup Arduino...   
Suggested by Martino Facchin on Arduino issue, that there are issues with jsons and uninstalling or going to older versions... Best to delete extracted stuff and reinstall.  

I have tried. At the time Arduino 1.8.7 and 1.8.6 would not start, but 1.8.5 does.  However when it tries to download, it now complains, that the stream is not a BZIP2 file... 
```
 Stream is not in the BZip2 format
java.lang.RuntimeException: java.io.IOException: Stream is not in the BZip2 format
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Stream is not in the BZip2 format
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.init(BZip2CompressorInputStream.java:240)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.<init>(BZip2CompressorInputStream.java:132)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.<init>(BZip2CompressorInputStream.java:109)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:96)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:81)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:159)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Could not find boards.txt in C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2. Is it pre-1.5?
```

OpusK, 2018-09-17:
@KurtE,

During the compression process, I seem to have specified the option incorrectly.
I compressed it again and modified the checksum and file size.

I tested it myself and found that it is installed well without BZip2 error.

I apologize again for any inconvenience.

OpusK, 2018-09-17:
Sorry @KurtE ,

It is my fault completely.
The compression included the parent folder, which meant that the Arduino packages folder path became deeper so that boards.txt could not be found.

Anyway, this problem has been fixed and I have confirmed that it works with 1.8.7.
Thank you again for your report.

KurtE, 2018-09-17:
Thanks, I downloaded updated version and it appears to work for me as well!

closed 2018-09-18 by OpusK