ROBOTIS-GIT/OpenCR #131: DMA and Asynchrnous Uarts...

KurtE, 2018-09-18:
Not sure if I should create this as a new issue (discussion) or as part of the 
Asynchronous Serial issue/PR posts for OpenCM:
https://github.com/ROBOTIS-GIT/OpenCM9.04/pull/38
https://github.com/ROBOTIS-GIT/OpenCM9.04/issues/36

**Background**
As I mentioned in the above issues, I have a simpler update to the system to support Asynchronous Serial access than the one in the Pull Request.  Currently up on my Fork/Branch: https://github.com/KurtE/OpenCR/tree/Async_Serial_minimal

I have it working reasonably using Interrupts, that is if I call: 
```
HAL_StatusTypeDef drv_uart_write_it(uint8_t uart_num, const uint8_t *wr_data, uint16_t Size)
{
  return HAL_UART_Transmit_IT(&huart[uart_num], (uint8_t *)wr_data, Size);  
}
```
Which is my wrapper function which I call with the portion of the tx_buffer of the UARTClass code (2K in size).  
It tries to write all of the bytes between iTail and iHead...

This is working.  So I thought I would try to enable DMA on Serial3 as you have it defined to use DMA.  I have a new wrapper: 
```
HAL_StatusTypeDef drv_uart_write_dma(uint8_t uart_num, const uint8_t *wr_data, uint16_t Size)
{
  return HAL_UART_Transmit_DMA(&huart[uart_num], (uint8_t *)wr_data, Size);  
}
```

Which I am debugging now.  In the UART code,  I have a wrapper function to decide which one to call:
```
void inline UARTClass::startNextTransmitDMAorIT()
{
    tx_write_size = (tx_buffer.iTail < tx_buffer.iHead)? tx_buffer.iHead-tx_buffer.iTail : SERIAL_BUFFER_SIZE-tx_buffer.iTail;
    if ( _uart_mode == DRV_UART_IRQ_MODE) {
      if (drv_uart_write_it(_uart_num, &tx_buffer.buffer[tx_buffer.iTail], tx_write_size) != HAL_OK) 
      {
        tx_write_size = 0;  // error so clear it out
      }
    }
    else
    {
      if (drv_uart_write_dma(_uart_num, &tx_buffer.buffer[tx_buffer.iTail], tx_write_size) != HAL_OK)
      {
        Serial.println("Serial DMA write fail");
        tx_write_size = 0;  // error so clear it out
      }
    }
}
```
To use this I had to also make changes to initialize the DMA structure in void HAL_UART_MspInit(UART_HandleTypeDef* huart),  I am only doing this currently for Serial3 and only if  is_uart_mode is DMA... (Had to move assignment up earlier...).... 

**Now for the issues...** 
1) Currently I am getting some random garbage out using DMA.  As I showed earlier should be getting the same parameters to HAL_UART_Transmit_DMA as is passed to HAL_UART_Transmit_IT and the IT was working... 

Current wondering if the DMA requires the buffer passed in to be aligned probably to 32 bit alignment? 

2) How to deal with multiple possible DMA Channels that may want the same DMA stream. 
Real case example:   

UART3_TX can either use: DMA1_Stream3 (Channel 4) or DMA1_Stream4(Channel 7), but we have already setup to optionally allow SPI2 to use DMA mode and SPI2 uses both Stream3(channel) and Stream4(channel 0).  

So how do we setup the code such that majority of programs use DMA on Serial3, UNLESS the SPI2 wants it...   First thing you run into.  Is for now assume I choose DMA1_Stream3.  Both the Uart code and the SPI code are going to want to define the function:  DMA1_Stream3_IRQHandler, and you can not have both, do this or linker will complain... 

But, all these Dma Stream functions do is something like:
```
void DMA1_Stream3_IRQHandler(void)
{
  HAL_DMA_IRQHandler(huart[DRV_UART_NUM_3].hdmatx);
}
```
Likewise in the SPI code again calls same handler with different hdma handle... 

So wondering if we could define simple HDMA handle setup.  Maybe something like: 
```
DMA_HandleTypeDef *DMA1_Streams_hdma[8];
DMA_HandleTypeDef *DMA2_Streams_hdma[8];
```
And then define a bunch of boiler plate functions like:
```
```
void DMA1_Stream3_IRQHandler(void)
{
  HAL_DMA_IRQHandler(DMA1_Streams_hdma[3]);
}
```
And then probably have wrapper functions that set this array, when appropriate.  Maybe also a simple function that checks to see if we own it or not...   In Serial3 case probably default to use DMA, unless SPI2 asks for it and then revert to use IT functions instead. 

The assignment function might have to detect if the stream is currently active or not and maybe optionally somehow wait until DMA has completed before assignment.

There may be other issues, but this is how far I have gotten so far.  Suggestions?

Thanks
Kurt

KurtE, 2018-09-18:
In regards to issue 1) I am guessing that it is memory alignment issue. 

That is I tried:
```
// BUGBUG:: setup aligned buffer to see if DMA works better.
static uint8_t tx_buffer_aligned[SERIAL_BUFFER_SIZE] __attribute__ ((aligned (4))); // make sure dword aligned 
void inline UARTClass::startNextTransmitDMAorIT()
{
    tx_write_size = (tx_buffer.iTail < tx_buffer.iHead)? tx_buffer.iHead-tx_buffer.iTail : SERIAL_BUFFER_SIZE-tx_buffer.iTail;
    if ( _uart_mode == DRV_UART_IRQ_MODE) {
      if (drv_uart_write_it(_uart_num, &tx_buffer.buffer[tx_buffer.iTail], tx_write_size) != HAL_OK) 
      {
        tx_write_size = 0;  // error so clear it out
      }
    }
    else
    {
      memcpy(tx_buffer_aligned, &tx_buffer.buffer[tx_buffer.iTail], tx_write_size); // BUGBUG...
      if (drv_uart_write_dma(_uart_num, tx_buffer_aligned, tx_write_size) != HAL_OK)
      {
        Serial.println("Serial DMA write fail");
        tx_write_size = 0;  // error so clear it out
      }
    }
}
```
And now the outputs to The DXL is correct...  But I don't want to use this as solution, as having to copy memory partially removes benefit of DMA...  So may try converting to the two buffer solution...

KurtE, 2018-09-21:
Not sure if anyone might have suggestions?  I am running into an issue with the DMA version of the UART outputs... 

That is, In the function, that starts up the next output to UART, which can be either IT or DMA, The code is:
```
void inline UARTClass::startNextTransmitDMAorIT()
{
  uint32_t cbWrite;
  uint8_t *pb; 
  if (tx_buffer.pb <= tx_buffer.buffer2) 
  {
    // We were filling buffer1
    cbWrite = tx_buffer.cb;
    pb = tx_buffer.buffer1; 
    tx_buffer.pb = tx_buffer.buffer2;
    tx_buffer.cb = 0;
  }
  else 
  {
    cbWrite = tx_buffer.cb;
    pb = tx_buffer.buffer2; 
    tx_buffer.pb = tx_buffer.buffer1;
    tx_buffer.cb = 0;
  }
  tx_buffer.transfer_active = 1;
  if ( _uart_mode == DRV_UART_IRQ_MODE) {
    if (drv_uart_write_it(_uart_num, pb, cbWrite) != HAL_OK) 
    {
      tx_buffer.transfer_active = 0;  // clear as we may never get called again to clear
    }
  }
  else
  {
    // See if we can write this data out to Serial2 as well?
    Serial2.printf("%x: ", (uint32_t)pb);
    for (uint32_t i = 0; i < cbWrite; i++) Serial2.printf("%02x ", pb[i]);
      Serial2.println();
    if (drv_uart_write_dma(_uart_num, pb, cbWrite) != HAL_OK)
    {
      tx_buffer.transfer_active = 0;  // clear as we may never get called again to clear
    }
  }
}
```
The IT version appears to work just fine, but I am seeing some garbage in the DMA case, so I added the debug code above, which for the one DMA Serial port (3 which is DXL), I dump in hex to Serial2 the contents of the buffer that will be output to Serial3... The data is correct coming out of Serial2, but not Serial3... 
[image]
Maybe hard to see in above Logic Analyzer output, but Top row (DXL) looks like ping to FB 
As you can see in line 3 with my DXL logic analyzer... 2nd line is an IO pin I setup to mirror the direction pin...  
But the DMA output was: ff ff fd 00 fb 3 0 1 31 6
Where as the expected output and the one that went out to Serial2 (IT) was:
2001a85c: ff ff fd 00 fe 03 00 01 31 42

Suggestions?  Anyone run into anything like this?

Kurt

chcbaram, 2018-09-21:
Did you consider cache coherency when you use DMA for tx?
STM32F746 has cache and enabled in OpenCR so you have to consider when you use DMA. 

That's why UART rx buffer is located in none cacheable memory like this.

static uint8_t  drv_uart_rx_buf[DRV_UART_NUM_MAX][DRV_UART_RX_BUF_LENGTH] __attribute__((section(".NoneCacheableMem")));

Or you have to do write back before transferring data via DMA so RAM data must be same with cache memory.

And you need to use dma function for restarting.

Please check cache coherency first.

Thanks.

chcbaram, 2018-09-21:
For critical section, I guess you recommended right solution. 
I agree it could be added overhead when you use disable/enable global interrupt.

So I used buffer index with one word length ( uint32_t for ARM). It makes impossible that the data could be changed via other ISR.

I think anther option is that you use different index for ring buffer. ex) tx/rx.
When you write data to buffer, you use tx index and you use rx index for reading.

I hope I understood well what you said.

For DMA buffer aligne.
I think you don't need aligned buffer for DMA.
When I used DMA tx for SPI, there were no problem.

https://github.com/chcbaram/nucleo_audio/blob/master/nucleo_fw/src/hw/core/spi.c

This is the example spi tx dma for transferring large data.
But it was different MCU so it could be different for STM32F746.

Thanks.

KurtE, 2018-09-21:
Thanks, will try the .NoneCachableMem option! 
So if it does not need to be aligned, will maybe go back to circular buffer instead of two buffers... 

As for critical section, I am trying to avoid the issues.  Again with Head versus tail.. or buffer 1 and buffer 2, where who controls when it can shift.  Current stuff that I was experimenting with it is pretty much issolated, Except with the possible exception of if I higher priority interrupt comes in which then wants to write to the serial port.  Again this can work, but if queue gets full before all output, you need a way to either boast the Serial interrupt to higher priority or ... 

But first get the basics workign.

KurtE, 2018-09-22:
@chcbaram and @OpusK ,

The .NoneCachableMem option appears to have fixed it :) 

I created the TX buffers using that attribute.  I currently have the actual buffer created in Variant.cpp and pass it in to the constructor along with it's size.  So each Serial object can be tailored, depending on needs. 
Note: Currently they all are defined as 2K in size which is probably a complete overkill.

Note: the RX buffers for Serial look like there is 3K allocated.  1K for the DMA and 2K in the UartClass.  This is probably also way to much for most cases. 

I will probably do a PR with this soon, so you can compare this approach to the version I did in OpenCM.  My guess is you will like this version better. 

Before I do, I still need to handle DMA conflict between Serial3 and SPI2...  

Will create a DMA Handler class where both call to say that want the Stream and will probably try to set up that if SPI2 asks for it the TX for Serial3 will then go back to using IT mode, as probably majority of time Serial3 will be <= 1MBS and SPI2 might go 30MBS?... 

Thanks again

KurtE, 2018-09-22:
Forgot to mention: Current stuff is up on my Fork/Branch: 
https://github.com/KurtE/OpenCR/tree/Async_Serial_minimal

KurtE, 2018-09-25:
Following on to the previous post... 

There are issues with RX when transmit is non blocking.   I am playing around some.  I created a simple app, like we used to debug Serial issues on the Teensy 3.5 and Teensy 3.6 which have something like 6 uarts... 

I wire up a set of jumpers like:
Serial1.tx -> Serial2.rx
Serial2.tx -> serial4.rx
Serial4.tx -> Serial1.rx
Then run simple sketch, which receives data from Serial and then tries to forward the data over the different jumpers and when it receives data back on RX1 it writes it back out to Serial... 

Current sketch: 
```
uint32_t test_baud = 115200;
uint8_t buffer[1024]; // Serial buffer..
uint16_t cb;

void setup() {

  // put your setup code here, to run once:
  while (!Serial && millis() < 5000) ;
  Serial.begin(115200);
  Serial1.begin(test_baud);
  Serial2.begin(test_baud);
  Serial4.begin(test_baud);
  Serial.println("Serial ping pong test");
  pinMode(2, OUTPUT);
  digitalWriteFast(2, LOW);

}
void loop() {
  if (!Serial.available()) return; // don't have new line yet
  cb = 0;
  while (Serial.available()) {
    int ch = Serial.read();
    if (ch >= ' ') buffer[cb++] = ch;
  }
  buffer[cb] = 0; // null terminate
  Serial.printf("Line in: %s\n", buffer);
  Serial.print( "Echoed : ");

  uint16_t cbBack = 0; // how many bytes have made it through
  uint32_t start_time = micros();
  uint16_t cnt2 = 0;
  uint16_t cnt4 = 0;
  digitalWriteFast(2, HIGH);
  Serial1.write(buffer, cb); 
  while ((cbBack < cb) && ((micros()-start_time) < 1000000)) {
    if (Serial2.available()) {
      Serial2.write(Serial2.read());
      cnt2++;
    }

    if (Serial4.available()) {
      Serial4.write(Serial4.read());
      cnt4++;
    }

    if (Serial1.available()) {
      Serial.write(Serial1.read());
      cbBack++; // it made it through 
    }
  }
  digitalWriteFast(2, LOW);
  uint32_t delta_time = micros() - start_time;
  Serial.printf("\nDelta Time %dus, Cnts: %d %d %d %d time per char: %d psudo baud %d\n", delta_time, cb, cnt2, cnt4, cbBack,  delta_time/cb, 
      1000000/(cb * 10));
}
```
Currently if I run it on my current buffered stuff, it loses data along the way...
[image]
Output to debug terminal looks like:
```
Serial ping pong test
Line in: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789
Echoed : cdefghijklmnopqrstuvwxyz012345789
Delta Time 1000000us, Cnts: 62 60 60 33 time per char: 16129 psudo baud 1612
```

Note: it works OK on the current released code as things block as each tx goes out...
it's output looks like:
```Serial ping pong test
Line in: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789 
Echoed : ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789 
Delta Time 17458us, Cnts: 63 63 63 63 time per char: 277 psudo baud 1587
```
As you can see by screenshot there is no overlap of any outputs:
[image]

Next up see about using DMA input for all 4?

KurtE, 2018-10-17:
@OpusK,

Probably can close this one as it was merged in.

closed 2018-10-18 by OpusK