ROBOTIS-GIT/OpenCR #133: Unable to get the Open Manipulator with TB3 working in Hardware

Vincent7803, 2018-10-01:
Hi,
I am unable to get the turtlebot3 with open manipulator core running in the OpenCR board. I am using raspberry Pi and the open CR board. The set-up works well with Turtlebot3 waffle core. But the moment i flash the open manipulator core...i face issues. Unable to sync. I tried the  resolving the issue by following the following links...but no success...Can you pls help resolve this issue..
https://github.com/ROBOTIS-GIT/OpenCR/issues/126
https://github.com/ROBOTIS-GIT/OpenCR/issues/123
https://github.com/ROBOTIS-GIT/OpenCR/issues/122

Thanks
Vincent

Vincent7803, 2018-10-02:
Also in my set-up, I am using Rs-485 Dynamixel motors for (5 no's of XM430-W350-R for open manipulator) and (2 no's XM430-W210-T for waffle wheels). 

Thanks
Vincent

routiful, 2018-10-02:
Hi :) Sorry for late reply.

I did resolve the problem and push it in `develop` branch.
https://github.com/ROBOTIS-GIT/OpenCR/commit/dc825997581e0f845e9d9b17edbfd5d4e9ee3e82

I will merge it in `master` after one week due to attend IROS.

Vincent7803, 2018-10-03:
Thanks a ton Taehun....The issue is resolved. 

Regards
Vincent

routiful, 2018-10-03:
Great! If you have any further problem, please open issues :)

closed 2018-10-03 by routiful