ROBOTIS-GIT/OpenCR #134: Power breaker circuit of openCR

XiafeiWang, 2018-10-03:
Hi,
I am new of openCR. The output power is short circuit because of my misoperation, thus leading to the breaker circuit of power supply. Is there a overload protection switch on the board or any other methods to solve this question? Thank you

OpusK, 2018-10-04:
Hi, @XiafeiWang 

Can I see pictures of your product?
There is a fuse for power protection. 
[image]

Please check the status of fuse on the back of OpenCR.

XiafeiWang, 2018-10-04:
> Hi, @XiafeiWang
> 
> Can I see pictures of your product?
> There is a fuse for power protection.
> [image]
> 
> Please check the status of fuse on the back of OpenCR.

Yes, this's it. I have replaced a new one. Thank you!

closed 2018-10-04 by XiafeiWang