ROBOTIS-GIT/OpenCR #136: Dynamixel Workbench: finding out what are valid item names...

KurtE, 2018-10-08:
@OpusK @kijongGil @ROBOTIS-Will and others:

Probably should be opened against Dynamixel Workbench project, but issues often show up with programs written for OpenCR or OpenCM9.04...

There have been many issues raised up on the forum, like:
http://en.robotis.com/service/forum_view.php?bbs_no=2406990&page=1&save_sca=&sca=&save_stx=&stx=&sfl=

Where the users (including myself) either don't find or misread the names of the items they can read, for the different types of servos. 

Wish could simply have the user s_Monitor Dynamixel Workbench example app, and do a begin to start up, do a scan, and then do a command that prints out all of the valid string register names and current values. 

One way to resolve this is to allow access to the ControlTableItem list... My quick hack was to add to dynamixel_workbench.h (class header)
```
  // Wish List... 
  ControlTableItem* getControlItemPtr(uint8_t id);
  uint8_t getTheNumberOfItem(uint8_t id);
```

To dynamixel_workbench.cpp I added:
```
ControlTableItem* DynamixelWorkbench::getControlItemPtr(uint8_t id)
{
  return driver_.getControlItemPtr(id);
}

uint8_t DynamixelWorkbench::getTheNumberOfItem(uint8_t id)
{
  return driver_.getTheNumberOfItem(id);
}
```

Then to try it out I added an "info" command to the s_Monitor program I added a section to the processing of commands:
```
      if (cmd[0] == "info")
      {
        // Lets loop through all of the information for this servo...
        uint8_t id     = cmd[1].toInt();

        // first print model number stuff
      uint16_t model_number = 0;

        
        ControlTableItem *cti =  dxl_wb.getControlItemPtr(id);
        uint8_t cti_count = dxl_wb.getTheNumberOfItem(id);
        if (cti)
        {
          if (dxl_wb.ping(id, &model_number))
          {
            Serial.print("Servo Info for id : ");
            Serial.print(id, DEC);
            Serial.print(" model_number : ");
            Serial.println(model_number, DEC);
          }
          while (cti_count--)
          {
            Serial.print("  ");
            Serial.print(cti->item_name);
            Serial.print(": ");
            int32_t value = dxl_wb.itemRead(id, cti->item_name);
            Serial.println(value, DEC);
            cti++;
          }
        }
      }
```

Then I did a test run with two XL430-w250 servos:
```
-------------------------------------
Set portHandler Before scan or ping
-------------------------------------
help
begin  (BAUD)
scan   (RANGE)
ping   (ID)
info   (ID)
id     (ID) (NEW_ID)
baud   (ID) (NEW_BAUD)
torque (ID) (VALUE)
joint  (ID) (GOAL_POSITION)
wheel  (ID) (GOAL_VELOCITY)
write  (ID) (ADDRESS_NAME) (VALUE)
read   (ID) (ADDRESS_NAME)
reboot (ID) 
reset  (ID) 
-------------------------------------
[CMD] : begin

Succeed to begin(57600)
[CMD] : scan

Find 2 Dynamixels
ID : 1
ID : 2
[CMD] : info 1

Servo Info for id : 1 model_number : 1060
  ID: 1
  Baud_Rate: 1
  Operating_Mode: 3
  Torque_Enable: 0
  LED: 0
  Goal_Current: 0
  Goal_Velocity: 265
  Profile_Acceleration: 0
  Profile_Velocity: 0
  Goal_Position: 2045
  Moving: 0
  Present_Load: 0
  Present_Velocity: 0
  Present_Position: 2045
[CMD] : info 2

Servo Info for id : 2 model_number : 1060
  ID: 2
  Baud_Rate: 1
  Operating_Mode: 3
  Torque_Enable: 0
  LED: 0
  Goal_Current: 0
  Goal_Velocity: 265
  Profile_Acceleration: 0
  Profile_Velocity: 0
  Goal_Position: 2048
  Moving: 0
  Present_Load: 0
  Present_Velocity: 0
  Present_Position: 2048
```

Note: s_Monitor - has other issues which has bit me, with the parameter parsing... That is if you do the commands: 
```
begin
ping 1
scan
```
The system will only find servo 1 in the scan, as the cmd[1] will not be overwritten and so scan will see the parameter 1 from the ping above...   

In my version I have a few hacks which fix it,  the simplest is in split function to do something like:
````
 // Clear out anything in the first 4 strings...
  temp[0] = "";
  temp[1] = "";
  temp[2] = "";
  temp[3] = "";
```

Again unsure if this is something anyone would be interested in and/or which project... 
If not feel free to close...

KurtE, 2018-10-11:
Again sorry that this maybe was just a random idea...  

But in case you wish to take a look, I put the changes up at:
https://github.com/KurtE/OpenCR/tree/Workbench_get_servo_info

routiful, 2018-10-11:
Thank you for your contribution @KurtE  :)

Your idea is quite practical. I will add next update and inform it to you.

routiful, 2018-10-18:
This issue is closed after merge @KurtE branch to develop branch.

closed 2018-10-18 by routiful

KurtE, 2018-10-18:
@routiful and @OpusK,

Note: I had left this project with the first level of changes with sources that did not include the later stuff that I changed in OpenCM9.04.  I used the OpenCM9.04 version to main workbench as I liked it better and it helped reduce memory footprint.

But was waiting to verify if you liked it better or not... 

So this morning I pushed up the changes to this fork/branch that made this project the same as the other two..

But not sure if you can now pick these changes up from here.  Or should I sync up my develop branch, create new branch with these changes and request new PR?

Thanks
Kurt

KurtE, 2018-10-18:
@routiful 
Update: I went ahead and did some quick updates and new PR for both OpenCM9.04 to update the header files in the released folder, plus  update lib_f103.a...

I also did a PR for OpenCR to update the code to be the same as Dynamixel_Workbench project.