ROBOTIS-GIT/OpenCR #142: CRC error and update boot loader

mpena2099, 2018-10-22:
Hi!

I'm trying to update the boot loader of my OpenCR to eliminate the error "cmd_read_board_name fail : 0xF020" on upload some sketches.

First: I think the documentation is wrong. There is no XML on https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_bootloader/common/msg/xml/opencr_msg . I found it in https://github.com/ROBOTIS-GIT/OpenCR/blob/df6f6dcfacc6f400083210495a8f944728bf70d6/arduino/opencr_develop/opencr_bootloader/common/msg/xml/opencr_msg

Second: when I try to run the MavLink, I got:

[image]

No ideia what is wrong.

Thanks!
Mauricio

mpena2099, 2018-10-23:
Hi @OpusK .

I had already made the settings as specified in the manual. I checked again, redo the procedure and the error continues.

Almost all sketches are recorded correctly. Only a few of them not. :-/

Perhaps... USB cable problem?

OpusK, 2018-10-29:
Could you tell me which example you used?

If the code changes, the size of the Flash may be reduced, but is not the 0xF020 error you are seeing in the process of identifying the board name? (like "cmd_read_board_name fail : 0xF020")

mpena2099, 2018-10-29:
Hi @OpusK . My code is based on turtlebot3_core. 

That line (odometry_msg_pub.publish(&odometry_msg)) is to publish some Twist messages. If I comment it I can upload:
......................................................
flash_erase : 0 : 0.965000 sec
flash_write : 0 : 1.603000 sec 
CRC OK 10863C2 10863C2 0.005000 sec
[OK] Download 
jump_to_fw 
......................................................

But WITH that line...
......................................................
flash_erase : 0 : 0.917000 sec
flash_write : 0 : 1.612000 sec 
CRC Fail : 0xF020 : 10D16D4, 0 0.629000 sec
[FAIL] Download 
......................................................

Anyway, I can upload the turtlebot3_core with no problems.

OpusK, 2018-10-31:
@mpena2099,
Thanks for your report.

The last error occurred because the FLASH write was completed but the CRC check response timed out.
Therefore, if there is no problem with the flash itself, the firmware will work normally.

However, if these errors persist, we are going to debug in this regard.

mpena2099, 2018-10-31:
@OpusK , right now there is no CRC error...

 ......................................................
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 0.958000 sec
flash_write : 0 : 1.038000 sec 
CRC OK 10D7124 10D7124 0.006000 sec
[OK] Download 
jump_to_fw 
 ......................................................

No changes in code. Perhaps a different USB port, I don't know. :-/

But... OK, in future I'll ignore the CRC error if the firmware looks OK.

Thanks for your support!
Mauricio

closed 2018-10-31 by mpena2099