ROBOTIS-GIT/OpenCR #145: Can't find Dynamixel

routiful, 2018-10-26:
I tested `Find_Dynamixel` example of Dynamixel-Workbench and `usb_to_dxl` in develop branch.
But there are not proper working. I assume that it is caused by serial comm error.
I am debugging it but I can't find what is the main problem.

@KurtE  Can you please test it for me? Because recent PRs were almost merged came to your branch.

KurtE, 2018-10-26:
Hi @routiful and @OpusK,

I will try running them on my machine.  I did earlier especially the find_dynamixel.   Remember I actually included an updated on, in later changes, which allowed you to rerun the test. 

Might help to have additional information, like when you compiled, did you get any compiler warning? 
Assume you run with Arduino configured to show all warnings (setting under preferences dialog).

Also are there any messages at the ed of compiler, like using Library X from ... And not libary x from ...
In particular do you have a different copy of Dynamixel SDK in your Arduino library folder.   Remember there was a change to SDK to do a flush before it swtiched the IO pin to go back into dynamixel Read mode. 

If you were a normal user, I would have said, things like: Make sure you have power, servos plugged in, power switch turned on....  Check that servo is working... But again you are not normal users! 

Beyond that not sure...

KurtE, 2018-10-26:
@routiful and @OpusK and @kijongGil ,

I think I see the problem, I just updated my develop to the last change, which was the Sync DynamixelSDK, change.   With this change, I See there was changes to 

port_handler_arduino.cpp, that  in the changes: 

https://github.com/ROBOTIS-GIT/OpenCR/commit/185e36be58c4b33bd14d63ac163508aec9ce5fe1#diff-0533b3a0549dc5a2a34d48162397f443

You will see in the function: void PortHandlerArduino::setTxDisable()
The calls to flush were removed. 

Another change in this file I wondered about was changing the latency timer from 4 to 16?

KurtE, 2018-10-26:
You might try out PR: #146