ROBOTIS-GIT/OpenCR #147: Fail to ping Dynamixel

mclaubee, 2018-10-31:
Hi,
I'm having issue with the dynamixel comm on my OpenCR (new, just bought it a couple of weeks ago). It was working for a while but now it fails to communicate with the servos. I tried with DynamixelWorkbench (with Arduino) to ping the servos but it shows:

Start..
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!

I tested with MX106 and XM430-W350-R. Protocol (2), Baudrate (I changed it to 2M), ID (1), all checked. 
Could it be OpenCR hardware problem? Because it was working previously.

PS: I tested the IMU, LED, serial comm. all work fine. OpenCR's firmware is 1.3.2.

routiful, 2018-11-01:
Hi :)

Did you update MX-106-2 firmware? 
http://emanual.robotis.com/docs/en/dxl/mx/mx-106-2/

mclaubee, 2018-11-01:
Yes it is. Version: 42.

mclaubee, 2018-11-14:
No problem. I will try that, thanks. But it was running at 2M at the beginning, and the OP3's servos are running at 2M too.

routiful, 2018-11-14:
Then, I will test that. Please tell me which motors and how many.

mclaubee, 2018-11-14:
It was working with 2 MX-28, 4 MX-64 and 14 MX-106. Thank you.

routiful, 2018-11-15:
I restricted the number of dxl due to memory usage

https://github.com/ROBOTIS-GIT/OpenCR/blob/9da4ae9fd4506df6e068c2a6d381f268a0ba371a/arduino/opencr_arduino/opencr/libraries/DynamixelWorkbench/include/dynamixel_workbench_toolbox/dynamixel_tool.h#L42

You are supposed to change array size 16 to 20 or higher.

mclaubee, 2018-11-15:
It was actually working with the OP3 firmware. After I had the problem to initialise the servos then I tried to troubleshoot with the OpenCR workbench with just one servo (MX-106). I tried with 1M too but it doesn't work.

routiful, 2018-12-13:
I will update Dynamixel-Workbench library.
This version can handle upto 30 Dynamixels

https://github.com/ROBOTIS-GIT/OpenCR/blob/01006432bc9749a7512a703cbef3938424088df1/arduino/opencr_arduino/opencr/libraries/DynamixelWorkbench/include/dynamixel_workbench_toolbox/dynamixel_tool.h#L30

Please check it and if you have any further issue, re-open this thread

Thanks.

closed 2018-12-13 by routiful

abdul-mannan-khan, 2021-06-03:
I am having the same issue. It is not working straight forwardly as mentioned [here](https://www.youtube.com/watch?v=0_M0Da9SHDw&t=20s). Maybe I am doing something wrong.

ROBOTIS-Will, 2021-06-04:
@abdul-mannan-khan 
The most up to date example is written based on DYNAMIXEL with ID = 1, Baudrate = 57600, DYNAMIXEL Protocol Version = 2.0.
If you are using different configuration, you'll need to modify the example accordingly.

abdul-mannan-khan, 2021-06-04:
I solved it. Thank you.