ROBOTIS-GIT/OpenCR #150: 0xF020 OpenCR upload

pokitoz, 2018-11-16:
Hello,

I am trying to upload a code on the OpenCR, but I cannot because of the following error when uploading:

```
Arduino: 1.8.4 (Linux), Board: "OpenCR Board, OpenCR Bootloader"
...
An error occurred while uploading the sketch
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_383628/test.ino.bin 
file size : 170 KB
Open port OK
Clear Buffer Start
An error occurred while uploading the sketch
Clear Buffer End
>>
cmd_read_board_name fail : 0xF020
```

It is the first time I am seeing this error. I saw some issues on this Github that got it working but I am still stuck with this Timeout error.
Let me know what kind of logs you need if this is not enough. Here is what I tried:

- Change Arduino version. Now using 1.8.4
- Update Bootloader (DFU mode + reset)
- Update OpenCR version from 1.3.0 to 1.3.2
- Change USB cable
- Connect the USB directly to the Desktop (no hub)
- Followed the tutorial http://emanual.robotis.com/docs/en/parts/controller/opencr10

I tried uploading different design (almost all the examples in the TurtleBot3 Examples) but none of them worked.

pokitoz, 2018-11-19:
Hello @OpusK 

Thanks a lot. I completely missed this chapter in the instruction thinking it was the same as updating the boot-loader :face_with_head_bandage: 

Anyway, I could update the new code by going to the recovery mode :D 
Thanks again !

closed 2018-11-19 by pokitoz

ubuntuyin, 2019-04-15:
我看到上面的方法，固件恢复模式也有尝试，状态灯无变化，是否是没有成功进入，怎么查看？程序上传仍然显示cmd_read_board_name fial:0xF020.求解。